NicoHood/HoodLoader2 #2: Error in picture

bgma, 2015-01-09:
When I tested HID upptäkte I picture "Arduino-unomega-16u2-pinout" and my "Arduino UNO R3 / Arduino-Compatible 'does not match with each other.
The difference is the 4 pins PB4, PB5, PB6 and PB7 are exchanged.

/BG

NicoHood, 2015-01-09:
may you send me a simple picture of your pin order? It is possible that unofficial board have a different layout since the 16u2 is normally not concerned to be used differently.

bgma, 2015-01-09:
because I have both "UNO R3" and "UNO Rev 3" I have compared the circuit boards and what I can see they are equal. I have also checked the wiring diagram from "arduino.cc" and what I can come up with, they are shifted. I have tried using "HID_Mouse" and "Arduino UNO R3 / Arduino-Compatible" to change "pinButtonT" to 4,5,6,7 and then it looks as follows:
Picture PB4 = R3 PB5
Picture PB5 = R3 PB7
Picture PB6 = R3 PB4
Picture PB7 = R3 PB6
I have not checked the other signal names (eg 18, PCINT4, T1) for these sticks just PBN.
[image]
My UNO R3

[image]
Pictue of UNO R3

[image]
Pictue of UNO R3 JP2 SPI

[image]
Pictue of UNO Rev 3

[image]
Pictue of UNO Rev 3 JP2 SPI

NicoHood, 2015-01-09:
bgma, thx for the fotos, you are right. I never checked that back. The photo wasnt originally made by me. I will ask for the change or edit it myself with paint.net next time. Thank you very much.

NicoHood, 2015-01-09:
Does this look correct to you? (its currently on the dev branch, pics take some time to load)
https://github.com/NicoHood/HoodLoader2/commit/1731024fcd851b0cc61478d339b694b26e75cae1

bgma, 2015-01-09:
Yes, it looking good.
 
Date: Fri, 9 Jan 2015 14:16:50 -0800

1731024

  
    
    
  
  
 		 	   		  =

NicoHood, 2015-01-09:
Okay. So it seems you are using the HID 2.1 with HoodLoader2. Do you think the new HID version is ready to publish? I have to recompile the HL2 2.0.3 and do some minor stuff but overall I finished for this release. BTW: Today i released a few updates. Not sure if you really have the lastest version. Installation is now even more easy. What do you think?

You may also close the issue here if the problem is solved and contact me on my blog if you have anything else to comment/ask/say. Thanks :)

bgma, 2015-01-09:
I'll download everything and start from scratch to test.
 
Date: Fri, 9 Jan 2015 15:07:24 -0800

bgma, 2015-01-10:
Ok, It seems to be working!

closed 2015-01-10 by bgma

NicoHood, 2015-01-10:
One thing i want to notice here as well:

I have got a board with a 32u2 as usb chip. This board has the ISP header rotated 180 degrees. The 4 pinheader is placed normally.

So be carefull with non original boards.

bgma, 2015-01-10:
Evil !!!!
Think that it is a bit lazily by the constructor that does not stick to the original.

bgma, 2015-01-10:
If you have not thought of it myself, it would be good with a small item in the setup steps?

NicoHood, 2015-01-10:
What do you mean? Do you mean I should add a little notice that the ISP header could be flipped on non official boards?

bgma, 2015-01-10:
was thinking of how I work sometimes, especially if you have something that stands out, it's easy to overlook the details. if you do not like the idea, ignore it.
 
Date: Sat, 10 Jan 2015 11:17:43 -0800

NicoHood, 2015-01-10:
Added. Installation should be clear now. Its hard to explain everything you need but not go too much into detail. If you wish to change anything else, let me know ;)

bgma, 2015-01-11:
Crystal clear!
Thank you.

Have some thoughts that I do not know if I detangles without help, is not so good at C after being "talked Basic / VB / VBA" for about 20 years. my question is it possible to do the following?

The mouse on the right is later a writing pad with usb.
In computer do I need to access the mouse, keyboard and COMn.
At 328, I need access to ports & pins (No problem).

[image]
/Grateful for answers BG

bgma, 2015-01-11:
Wrong Place for this question?

NicoHood, 2015-01-11:
Yeah, better ask here and no longer on github issues:
https://nicohood.wordpress.com/about/

It will be possible soon. I am working on a 328-16u2 link. A protocol that packs the usb reports into a serial protocol and the 16u2 translates this into usb. As an alternative you could upload HoodLoader1 ( a firmware, no bootloader) via avr dude and get the same effect. Just that i will redo this work and improve it even more. I also plan on fixed firmwares with no cdc serial support to get better recognition on smartphones, consoles, gamepads in general (made a lot of problems). Thats the plan :)