NicoHood/HoodLoader2 #3: 6u2 to burn atmel bootloader

bogomips, 2015-01-19:
Hello, I read with interest and I'm wondering if is possible to use the 16u2 to burn a traditional atmega328p bootloader. It would be amazing, I mean to use 16u2 as ISP.
thanks

NicoHood, 2015-01-19:
Yes it is. Upload the HoodLoader1 as firmware via avrdude and read this section:
https://github.com/NicoHood/Hoodloader#16u2-as-isp-usage

The standard Arduino IDE sketch doesnt work for some reason. I have to fiddle into that when I have more time to do so. But the Hoodloader1 would work. You can also flash it even when no HL2 is installed. Then you have to use flip via dfu. Just try it :)

bogomips, 2015-01-19:
thanks! 
I had read that section, the doubt was born becouse I didn't find same section in HL2. So just the 1 has this feature! 
I'll try it.

NicoHood, 2015-01-19:
Yep. Trying to fix that sooner or later

NicoHood, 2015-01-24:
Fixed! Available with the HID Project:
https://github.com/NicoHood/HID/commit/4775b3881c23e4f6b9cd82fd101dfd980e3fe1b8

https://github.com/NicoHood/HID/wiki/Arduino-as-ISP

Please give me feedback if it also works for you.

closed 2015-01-24 by NicoHood