NicoHood/HoodLoader2 #6: Problem with 8u2 (uno)

thesymbol, 2015-03-23:
I have been trying to upload the HoodLoader2_NoUSB_Blink to the 8u2 (since i do not have the space to fit the HID library for keyboard). but every time it fails to verify. here is the output from Arduino 1.6.1 

    Sketch uses 646 bytes (15%) of program storage space. Maximum is 4 096 bytes.
    Global variables use 9 bytes (1%) of dynamic memory, leaving 503 bytes for local variables. Maximum is 512 bytes.
    Forcing reset using 1200bps open/close on port COM14
    PORTS {COM1, COM14, } / {COM1, COM14, } => {}
    PORTS {COM1, COM14, } / {COM1, COM14, } => {}
    PORTS {COM1, COM14, } / {COM1, COM14, } => {}
    PORTS {COM1, COM14, } / {COM1, COM14, } => {}
    PORTS {COM1, COM14, } / {COM1, COM14, } => {}
    PORTS {COM1, COM14, } / {COM1, COM14, } => {}
    PORTS {COM1, COM14, } / {COM1, COM14, } => {}
    PORTS {COM1, COM14, } / {COM1, COM14, } => {}
    PORTS {COM1, COM14, } / {COM1, COM14, } => {}
    PORTS {COM1, COM14, } / {COM1, COM14, } => {}
    PORTS {COM1, COM14, } / {COM1, COM14, } => {}
    PORTS {COM1, COM14, } / {COM1, COM14, } => {}
    PORTS {COM1, COM14, } / {COM1, COM14, } => {}
    PORTS {COM1, COM14, } / {COM1, COM14, } => {}
    PORTS {COM1, COM14, } / {COM1, COM14, } => {}
    PORTS {COM1, COM14, } / {COM1, COM14, } => {}
    PORTS {COM1, COM14, } / {COM1, COM14, } => {}
    PORTS {COM1, COM14, } / {COM1, COM14, } => {}
    PORTS {COM1, COM14, } / {COM1, COM14, } => {}
    PORTS {COM1, COM14, } / {COM1, COM14, } => {}
    Uploading using selected port: COM14
    C:\Program Files (x86)\Arduino/hardware/tools/avr/bin/avrdude -CC:\Program Files (x86)\Arduino/hardware/tools/avr/etc/avrdude.conf -v -patmega8u2 -cavr109 -PCOM14 -b57600 -D -Uflash:w:C:\Users\Marcus\AppData\Local\Temp\build3428156613028146164.tmp/HoodLoader2_NoUSB_Blink.cpp.hex:i 
    
    avrdude: Version 6.0.1, compiled on Jan 15 2015 at 16:58:43
             Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
             Copyright (c) 2007-2009 Joerg Wunsch
    
             System wide configuration file is "C:\Program Files (x86)\Arduino/hardware/tools/avr/etc/avrdude.conf"
    
             Using Port                    : COM14
             Using Programmer              : avr109
             Overriding Baud Rate          : 57600
             AVR Part                      : ATmega8U2
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PC6
             RESET disposition             : possible i/o
             RETRY pulse                   : SCK
             serial program mode           : yes
             parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             ByteDelay                     : 0
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :
    
                                      Block Poll               Page                       Polled
               Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom        65    20     4    0 no        512    4    128  9000  9000 0x00 0x00
               flash         65     6    64    0 yes      8192   64    128  4500  4500 0x00 0x00
               lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
               hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
               efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
               lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
               calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
               signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
    
             Programmer Type : butterfly
             Description     : Atmel AppNote AVR109 Boot Loader
    
    Connecting to programmer: .
    Found programmer: Id = "HL2.0.4"; type = S
        Software Version = 1.0; No Hardware Version given.
    Programmer supports auto addr increment.
    Programmer supports buffered memory access with buffersize=128 bytes.
    
    Programmer supports the following devices:
        Device code: 0x44
    
    avrdude: devcode selected: 0x44
    avrdude: AVR device initialized and ready to accept instructions
    
    Reading | ################################################## | 100% 0.00s
    
    avrdude: Device signature = 0x1e9389
    avrdude: reading input file "C:\Users\Marcus\AppData\Local\Temp\build3428156613028146164.tmp/HoodLoader2_NoUSB_Blink.cpp.hex"
    avrdude: writing flash (646 bytes):
    
    Writing | ################################################## | 100% 0.02s
    
    avrdude: 646 bytes of flash written
    avrdude: verifying flash memory against C:\Users\Marcus\AppData\Local\Temp\build3428156613028146164.tmp/HoodLoader2_NoUSB_Blink.cpp.hex:
    avrdude: load data flash data from input file C:\Users\Marcus\AppData\Local\Temp\build3428156613028146164.tmp/HoodLoader2_NoUSB_Blink.cpp.hex:
    avrdude: input file C:\Users\Marcus\AppData\Local\Temp\build3428156613028146164.tmp/HoodLoader2_NoUSB_Blink.cpp.hex contains 646 bytes
    avrdude: reading on-chip flash data:
    
    Reading | ################################################## | 100% 0.02s
    
    avrdude: verifying ...
    avrdude: verification error, first mismatch at byte 0x0000
             0x11 != 0x0c
    avrdude: verification error; content mismatch
    
    avrdude done.  Thank you."

NicoHood, 2015-03-23:
do you use the newest 2.0.4? Have you installed the 8u2 version? the 
16u2 version will not work on an 8u2.
I never tested it since i dont have a 8u2.

Does the usb serial still work to flash the 328?

Nico

thesymbol, 2015-03-23:
Yes im using the 2.0.4 version for the 8u2, the flashing went through without any errors, (i used a Arduino Mega ADK to flash the UNO with Hoodloader 2.0.4)

And yes i can still upload sketches to the 328. (for example the blink from arduinos example sketches).

i used these settings in the ino sketch:

    // make some of these false to reduce compile size (the ones you don't want)
    #define USE_ATMEGA8U2 true
    #define USE_ATMEGA16U2 false
    #define USE_ATMEGA32U2 false
    #define USE_AT90USB162 false // todo, not supported
    
    // also use Mega 16u2 bootloader files?
    #define USE_MEGA_BOOTLOADER false
    
    // force Mega bootloader for blind pin selection/DFU hex file
    #define FORCE_MEGA_BOOTLOADER false
    
    // DFU options (16u2 only)
    #ifdef USE_ATMEGA16U2
    #define USE_ATMEGA16U2_DFU true
    // use the full 16kb DFU Bootloader with USB-Serial
    // you have to deactuivate the mega bootloader then
    // but you can stull burn a mega DFU file in the followed setting
    #define USE_ATMEGA16U2_DFU_FULL false
    #endif

NicoHood, 2015-03-23:
Interesting. Try the UNO or Mega file. Both should work the same, no 
matter what board is used.
Can you try to upload a hex file via avrdude in the tools folder? the 
usbserial firmware for example.

I'd have to get a board myself to test this. and I wont buy this myself.

Nico

NicoHood, 2015-03-23:
Do you have a 16u2 to test this? Maybe the fuses are set wrong :S
I'd have to retest this myself. Maybe i uploaded crap?

thesymbol, 2015-03-23:
> Interesting. Try the UNO or Mega file. Both should work the same, no
matter what board is used.

what files are you reffering to?

> Can you try to upload a hex file via avrdude in the tools folder? the
usbserial firmware for example.

Getting the same error, or atleast a similar validation error.

> Do you have a 16u2 to test this? Maybe the fuses are set wrong :S
I'd have to retest this myself. Maybe i uploaded crap?

No i do not have a 16u2 to test, i wish i had. :(

NicoHood, 2015-03-23:
Too bad the ADK has no 16u2.
In the uploading process you can say 'M' for mega and 'U' for uno. try both.

Otherwise I have no clue why that error occurs. Is it a real atmega 8u2 
or atusb9082? What does the print on the chip say?

Where do you live? I'd be willing to test this if you send me the board 
on your cost or buy me one. ADK is 25€ on banggood.
Otherwise I suggest you to use a mega and buy a shield separate for usb 
host if needed.

Nico

thesymbol, 2015-03-23:
it says
ATMEL
mega8u2
and two more rows that i cant read.

i think the uno is a rev 1 since the mega8u2 chip is turned a bit so it looks like a diamond when the arduino is correctly orianted.

I think the mega has a 16u2 but cant make out the text that well, but i wont flash that board since im borrowing it from a friend.

Im not willing to pay for a board tho, i might just try and get my hands on a 32u2 or 16u2 uno later down the road.

NicoHood, 2015-03-23:
yeah. just grab a new one.
It would be interesting to get it working though. very weird.  And i 
have no real idea why. maybe I compiled the files wrong?

Can you please open an issue on github? maybe others have this problem 
as well. also as reminder for me.
I will try to recompile the bootloader. Just remind me. I think I 
probably forgot to change the size of the bootloader.

Nico

thesymbol, 2015-03-23:
this conversation is already on github ;)

Tried to flash an old hid .hex firmware from another project that i used to be able to flash and got this

    avrdude.exe: verifying ...
    avrdude.exe: verification error, first mismatch at byte 0x0000
                 0xff != 0x9a
    avrdude.exe: verification error; content mismatch
    
    avrdude.exe: safemode: Fuses OK (E:FC, H:D8, L:EF)
    
    avrdude.exe done.  Thank you.

here is the output from flashing the normal Arduino-usbserial-uno.hex

    avrdude.exe: verifying ...
    avrdude.exe: verification error, first mismatch at byte 0x0000
                 0xff != 0x9c
    avrdude.exe: verification error; content mismatch
    
    avrdude.exe: safemode: Fuses OK (E:FC, H:D8, L:EF)
    
    avrdude.exe done.  Thank you.

If i take your HoodLoader2_0_4_Uno_8u2.hex file and flash it it goes through and verify correctly

    avrdude.exe: verifying ...
	avrdude.exe: 8192 bytes of flash verified

	avrdude.exe: safemode: Fuses OK (E:FC, H:D8, L:EF)

	avrdude.exe done.  Thank you.

NicoHood, 2015-03-23:
how do you flash? via real ISP or via installation sketch?

Totally missed, that this was a github post. though it was by mail from my blog. well.... :D

thesymbol, 2015-03-23:
I dont have a real ISP, and im flashing with avrdude.exe from the tools folder directly to the UNO.
i also tried doing the bootloader_converter.lua way but the .hex file's turned into 16384 when the 8u2 only has 8192 of memory to flash to and that will just end badly if i try to flash that.

hehe, the more you know eh? ;)

thesymbol, 2015-03-23:
i tried the M way aswell nothing changed.

thesymbol, 2015-03-24:
Got some good news, i managed to manually upload the sketch (HoodLoader2_NoUSB_Blink) to the 8u2, but to do so i had to use avrdude and specify at90usb82 instead of atmega8u2 for the -p parameter. im guessing this is because its a rev 1 uno.

The command i used was this:
avrdude -C avrdude.conf -F -p at90usb82 -c avr109 -b 57600 -P COM14 -U flash:w:HoodLoader2_NoUSB_Blink.cpp.hex:i

now the RX and TX leds are blinking as they should :D

NicoHood, 2015-03-24:
Try to change this line to at90usb82, restart the IDE and try to upload.
It should then work. If yes, I'll probably add this MCU type as well.

Still weird though.

https://github.com/NicoHood/HoodLoader2/blob/master/avr/boards.txt#L164

thesymbol, 2015-03-24:
Already one step ahead of you, did that already (but copied the whole 8u2 section and renamed it to at90usb82).
But i get an signature verification error if i dont specify -F to skip the check.

    avrdude: Device signature = 0x1e9389
    avrdude: Expected signature for AT90USB82 is 1E 93 82
             Double check chip, or use -F to override this check.

NicoHood, 2015-03-24:
that is weird.

So you get an error with the IDE on both 8u2 and at90usb82.
With avrdude you only get an error with 8u2, right?

I am wondering why avrdude accepts the wrong device signature with manual control
A good idea is to provide a blink.hey file. good idea!

The only thing I can think of is to burn the hex file via ISP and not with the installation sketch. maybe the conversion was wrong.
I know this page has to be improved, but you may want to try ISP with one of your other arduinos.
https://github.com/NicoHood/HID/wiki/Arduino-as-ISP

thesymbol, 2015-03-24:
> So you get an error with the IDE on both 8u2 and at90usb82.
With avrdude you only get an error with 8u2, right?

in the IDE i get the signature verification check error (and it suggest to force it with -F)
so when i enter the same command manually into cmd with the -F parameter the upload goes through successfully.

> I am wondering why avrdude accepts the wrong device signature with
> manual control A good idea is to provide a blink.hey file. good idea!

The -F parameter makes avrdude skip the verification completely and just uploads the data to the 8u2 without questioning the signature of the device.

> The only thing I can think of is to burn the hex file via ISP and not with the installation sketch. maybe the conversion was wrong.
I know this page has to be improved, but you may want to try ISP with one of your other arduinos.
https://github.com/NicoHood/HID/wiki/Arduino-as-ISP

Doesnt there exist a list of devices (atmega16u2 etc) with their device signatures? could help in finding why the at90usb82 doesnt match the uno 8u2 chip.

NicoHood, 2015-03-24:
here is the signature
https://github.com/nickgammon/arduino_sketches/blob/master/Atmega_Board_Programmer/Atmega_Board_Programmer.ino#L282-L286

try to burn via isp, maybe the installation sketch did something wrong.

thesymbol, 2015-03-24:
still no success, still getting the verification error of the device if i use at90usb82 and with atmega8u2 i get the 0xff != 0x0c

NicoHood, 2015-03-25:
With an ISP you can do 2 things:
Burn a bootloader and try to upload via bootloader
or flash a program directly

Can you please try both versions? At least uploading the blink via the IDE via programmer should work, since it has nothing to do with the bootloader itself.

thesymbol, 2015-03-25:
Uploading normal ino for the 328 works.
I do not have the ADK anymore (had to give it back today) so i cannot test this if i need the ADK as ISP.

but if i do not can you point me in the right direction (im a bit confused by you phrasing) what do you mean by flashing a program directly?

NicoHood, 2015-03-26:
You can flash a program via bootloader or via ISP. You can burn Bootloaders only via ISP.

I want to know if its an IDE or chip problem or a bootloader problem. When you upload the blink via ISP there is no bootloader used. If this works, we know that its not a chip/IDE problem.

Then you can burn the bootloader via ISP instead of the installation sketch (which emulates an ISP but a bit different stil, with file conversion etc). Then we know if its an installation sketch problem or a real bootloader problem.

Just buy a new Arduino with a 16u2. One from china is 6,5€ so why not buy one with even a better usb chip? It takes some time till it arrives but otherwise you wont be able to use it, it seems. And I have no idea where to get a 8u2 from and also it doesnt really make sense to buy that "old" chip.

thesymbol, 2015-03-26:
Im going to buy one when i get the time for it, but im probably going to buy a real arduino.
 
I dont mind paying extra for the real deal :P
i will just have to make sure its a 16u2 and not a 8u2 but i guess all rev 3 has 16u2?

i will report back once i get my hands on a new arduino.

NicoHood, 2015-03-26:
All rev3 (official ones) have a 16u2 so you are at the secure side. Good idea to support the developers!

FYI: march 28th is Arduino day. I think a lot of shops offer sales. So you should have a look then (this saturday). Sparkfun, Adafruit and Seeed looks promising (for international shops, also see local sellers).

thesymbol, 2015-03-26:
Ah thx, then i dont have to worry that i get a different chip. I love the arduino so its a no brainer to support the developers :P

Ok I will look for deals on the 28th then :D

chandran13, 2015-04-12:
Greetings

I am chandran from Chennai India  I get the following error can you pl help not very good on electronis or programming

Arduino: 1.6.3 (Windows 7), Board: "Arduino Uno"

HoodLoader2_Installation_Sketch.ino.ino:218:35: fatal error: bootloader_atmega16u2.h: No such file or directory

compilation terminated.

Error compiling.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

regards
chandran

NicoHood, 2015-04-12:
What chip do you want to use? 8u2 or 16u2?
Have you changed anything in the sketch?

It compiles for me, not sure what you are doing wrong. In the IDE you must see several tabs with different .h files.

chandran13, 2015-04-13:
16u2

I have not touched any thing in sketch

NicoHood, 2015-04-13:
How did you open the file? via double click? Have you extracted the zip? May you take a photo of the opened IDE while the sketch is loaded? It seems the other .h files cannot be found. Try to redownload maybe.

NicoHood, 2015-04-15:
@thesymbol Can you please try to flash the Arduino via ISP? Also with a 2nd Arduino this is possible. I found some signature bugs in the installation sketch, maybe this was the problem.

thesymbol, 2015-04-16:
I will try once i get hold on an arduino, there were no deals at my local store during arduino day. But i will try and get hold of one soon to test flashing the arduino with another arduino as ISP.
Might go to my local store and pick one up today. If i get the time to.

NicoHood, 2015-08-08:
Can you please try again with the newest 2.0.5 version? (fast-usbserial branch)

thesymbol, 2015-08-08:
I still have not gotten around to buy a new Arduino, i have bought a Raspberry Pi B instead, if you know a way to use the raspberry pi to program the arduino i can check and see if its working. since i dont think my arduino has hoodloader on it anymore.

NicoHood, 2015-08-08:
And your Arduino with a 8u2 you had to give away? If you still have it, it would be nice if you can test it. You dont need a new one, I know that the 16u2 (normally) works.
You can use the standalone sketch to burn the bootloader, no 2nd Arduino needed. The docs are also updated for 2.0.5 waiting for testing.

thesymbol, 2015-08-08:
okay, i will test then (i still got the 8u2, it was the mega that i had to give back).

NicoHood, 2015-08-08:
That'd be great! Make sure to select the 8u2 hexfile in the Installation sketch.
If something is unclear in the docs, please report it. I changed a few things.

If the bootloader is instaled try to upload code to the IO MCU (328) like you are used to. If this works its all fine. You might need to verify if the bootloader is really installed, links all in the wiki.

thesymbol, 2015-08-08:
I actually have one question with the wiering, i do not have a 100nF cap at home, is it possible to do this without it or with a larger/smaller cap?

NicoHood, 2015-08-08:
All I know is you need it. You can try 10nF or 200nF. It should be something around that. Its just to remove the reset signal from the 16u2, to not reset the 328 while it burns the bootloader.

What kind of caps do you have? I'd say larger would work better. But dont use Elkos.

thesymbol, 2015-08-08:
100uF and 2.2uF mostly caps i pulled out from old circuits that aint in use anymore.

thesymbol, 2015-08-08:
found a 33nF cap 3.3v will try with that.

NicoHood, 2015-08-08:
If you have more than one, try them in parallel.

thesymbol, 2015-08-08:
sadly i do not have anymore only found one.

NicoHood, 2015-08-08:
Does it work?

thesymbol, 2015-08-08:
it did not work with 33nF, it says Attempting to enter programming mode ... and then the led flashes constantly  (the L led).

thesymbol, 2015-08-08:
do you think a 10uF would work?

NicoHood, 2015-08-08:
Wait. Did you use the 2.0.5 version? (not the master!)
https://github.com/NicoHood/HoodLoader2/tree/fast-usbserial

I dont know if its not clear enough, but you wont see any message "programming finished". A fast blinking leds indicates success, slow blinking means error. (while programming it will also blink fast, so you need to wait some time).

thesymbol, 2015-08-08:
yes i did download the fast-usbserial scetch and changed the #define HEXFILE 3 to 9 (for the Uno 8u2).

i have left it on for a good 2min now, or wait i just checked the machine id and it says REV_0205.
my bad should have checked there first.

NicoHood, 2015-08-08:
great!
You now may want to test uploading to the 328 and to the 8u2 if wanted

You need to manually install the hardware file for the 8u2, no boards manager solution is published yet. The old version wont work. In the HID project is an example how to get back to bootloader again if you want to permanently remove the 8u2 sketch again.

Was the instruction clear enough in the docs? Any suggestions?

thesymbol, 2015-08-08:
Should i follow these instructions for the hardware files or what do you mean with manually installing it?

Instructions are good enough i think some suggestions:
I did not get at first what you ment by "Recovery Buttons were removed in 2.0.5, you can ignore this." since this wasnt mentioned anywhere during the instructions.

Would also be good to update the images with the new version or atleast put some text in to how to change settings in the guide, might help some that just look in the guide and not in the scetch.

NicoHood, 2015-08-08:
To upload a sketch to the 328 you can use the normal Arduino setting.
But since you now have HL2.0.5 installed you might want to use its features (upload programs to the USB MCU). Therefore you need the hardware defintions files. And you need to manually place them in your sketchbook (for 2.0.4 you can use the boards manager). then you are able to upload programs to the 8u2 as well.

thesymbol, 2015-08-08:
ok, will do.
im running windows 10 now, will the drivers still work for windows 10 (just follow the 8/8.1 instructions)?

NicoHood, 2015-08-08:
No drivers needed for win 10 (from what ive heard). Tell me if this works for you.

thesymbol, 2015-08-08:
I can flash the 328 no problem, but still getting the verification error on the 8u2.

```
Sketch uses 646 bytes (15%) of program storage space. Maximum is 4 096 bytes.
Global variables use 9 bytes (1%) of dynamic memory, leaving 503 bytes for local variables. Maximum is 512 bytes.
Forcing reset using 1200bps open/close on port COM14
PORTS {COM1, COM14, } / {COM1, COM14, } => {}
PORTS {COM1, COM14, } / {COM1, COM14, } => {}
PORTS {COM1, COM14, } / {COM1, COM14, } => {}
PORTS {COM1, COM14, } / {COM1, COM14, } => {}
PORTS {COM1, COM14, } / {COM1, COM14, } => {}
PORTS {COM1, COM14, } / {COM1, COM14, } => {}
PORTS {COM1, COM14, } / {COM1, COM14, } => {}
PORTS {COM1, COM14, } / {COM1, COM14, } => {}
PORTS {COM1, COM14, } / {COM1, COM14, } => {}
PORTS {COM1, COM14, } / {COM1, COM14, } => {}
PORTS {COM1, COM14, } / {COM1, COM14, } => {}
PORTS {COM1, COM14, } / {COM1, COM14, } => {}
PORTS {COM1, COM14, } / {COM1, COM14, } => {}
PORTS {COM1, COM14, } / {COM1, COM14, } => {}
PORTS {COM1, COM14, } / {COM1, COM14, } => {}
PORTS {COM1, COM14, } / {COM1, COM14, } => {}
PORTS {COM1, COM14, } / {COM1, COM14, } => {}
PORTS {COM1, COM14, } / {COM1, COM14, } => {}
PORTS {COM1, COM14, } / {COM1, COM14, } => {}
PORTS {COM1, COM14, } / {COM1, COM14, } => {}
Uploading using selected port: COM14
D:\Program Files (x86)\arduino-PR-3640-BUILD-375\hardware\tools\avr/bin/avrdude -CD:\Program Files (x86)\arduino-PR-3640-BUILD-375\hardware\tools\avr/etc/avrdude.conf -v -patmega8u2 -cavr109 -PCOM14 -b1200 -D -Uflash:w:C:\Users\THESYM~1\AppData\Local\Temp\build188950287897520202.tmp/sketch/HoodLoader2_NoUSB_Blink.cpp.hex:i 

avrdude: Version 6.0.1, compiled on Apr 15 2015 at 19:59:58
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "D:\Program Files (x86)\arduino-PR-3640-BUILD-375\hardware\tools\avr/etc/avrdude.conf"

         Using Port                    : COM14
         Using Programmer              : avr109
         Overriding Baud Rate          : 1200
         AVR Part                      : ATmega8U2
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC6
         RESET disposition             : possible i/o
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no        512    4    128  9000  9000 0x00 0x00
           flash         65     6    64    0 yes      8192   64    128  4500  4500 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : butterfly
         Description     : Atmel AppNote AVR109 Boot Loader

Connecting to programmer: .
Found programmer: Id = "HL2.0.5"; type = S
    Software Version = 1.0; No Hardware Version given.
Programmer supports auto addr increment.
Programmer supports buffered memory access with buffersize=128 bytes.

Programmer supports the following devices:
    Device code: 0x44

avrdude: devcode selected: 0x44
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9389
avrdude: reading input file "C:\Users\THESYM~1\AppData\Local\Temp\build188950287897520202.tmp/sketch/HoodLoader2_NoUSB_Blink.cpp.hex"
avrdude: writing flash (646 bytes):

Writing | ################################################## | 100% 0.11s

avrdude: 646 bytes of flash written
avrdude: verifying flash memory against C:\Users\THESYM~1\AppData\Local\Temp\build188950287897520202.tmp/sketch/HoodLoader2_NoUSB_Blink.cpp.hex:
avrdude: load data flash data from input file C:\Users\THESYM~1\AppData\Local\Temp\build188950287897520202.tmp/sketch/HoodLoader2_NoUSB_Blink.cpp.hex:
avrdude: input file C:\Users\THESYM~1\AppData\Local\Temp\build188950287897520202.tmp/sketch/HoodLoader2_NoUSB_Blink.cpp.hex contains 646 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.03s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x0000
         0xff != 0x0c
avrdude: verification error; content mismatch

avrdude done.  Thank you.
```

NicoHood, 2015-08-08:
Can you try to upload the blink example to the 8u2 with pin 17?
If the led blinks it indicated that the sketch can upload but only verifying fails for some reason.

You can also try to upload a firmware via avrdude (maybe the IDE setting is wrong).
https://github.com/NicoHood/HoodLoader2/wiki/Uploading-programs#flasherase-firmwareshex-files-with-avrdude

I suggest you this hex file which should work with a 4kb limit:
https://github.com/urjaman/fast-usbserial

The firmware acts as USB-Serial (so actually the same function as HL2 just as firmware, not bootloader).

thesymbol, 2015-08-08:
Still got the same problem as before, my chip is only working when i set it to model at90usb82 and forces it to write.
atmega8u2 does not seem to work in avrdude with my chip.

```
avrdude.exe -C avrdude.conf -v -F -p at90usb82 -c avr109 -P COM14 -b 1200 -D -U flash:w:HoodLoader2_NoUSB_Blink.cpp.hex:i
```
this command returns a success when writing the HoodLoader NoUSB Blink example.
getting this output:
```
avrdude: Version 6.0.1, compiled on Apr 15 2015 at 19:59:58
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "D:\Program Files (x86)\arduino-PR-36
40-BUILD-375\hardware\tools\avr/etc/avrdude.conf"

         Using Port                    : COM14
         Using Programmer              : avr109
         Overriding Baud Rate          : 1200
         AVR Part                      : AT90USB82
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC6
         RESET disposition             : possible i/o
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page
      Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  Max
W   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ---
-- ---------
           eeprom        65    20     4    0 no        512    4    128  9000  90
00 0x00 0x00
           flash         65     6   128    0 yes      8192  128     64  4500  45
00 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  90
00 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  90
00 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  90
00 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  90
00 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0
 0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0
 0 0x00 0x00

         Programmer Type : butterfly
         Description     : Atmel AppNote AVR109 Boot Loader

Connecting to programmer: .
Found programmer: Id = "HL2.0.5"; type = S
    Software Version = 1.0; No Hardware Version given.
Programmer supports auto addr increment.
Programmer supports buffered memory access with buffersize=128 bytes.

Programmer supports the following devices:
    Device code: 0x44
[... 98 lines cut ...]
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% -0.00s

avrdude: Device signature = 0x1e9389
avrdude: reading input file "C:\Users\THESYM~1\AppData\Local\Temp\build188950287897520202.tmp/sketch/HoodLoader2_NoUSB_Blink.cpp.hex"
avrdude: writing flash (646 bytes):

Writing | ################################################## | 100% 0.11s

avrdude: 646 bytes of flash written
avrdude: verifying flash memory against C:\Users\THESYM~1\AppData\Local\Temp\build188950287897520202.tmp/sketch/HoodLoader2_NoUSB_Blink.cpp.hex:
avrdude: load data flash data from input file C:\Users\THESYM~1\AppData\Local\Temp\build188950287897520202.tmp/sketch/HoodLoader2_NoUSB_Blink.cpp.hex:
avrdude: input file C:\Users\THESYM~1\AppData\Local\Temp\build188950287897520202.tmp/sketch/HoodLoader2_NoUSB_Blink.cpp.hex contains 646 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.03s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x0000
         0xff != 0x0c
avrdude: verification error; content mismatch

avrdude done.  Thank you.
```

NicoHood, 2015-08-08:
hm maybe you have an at90usb82 and this causes problems.
I will try to recompile the original lufa CDC bootloader for you in a few minutes.

Its possible that this is an avrdude bug.

thesymbol, 2015-08-08:
When i try to run the same at90usb82 command without -F i get this:

```
avrdude.exe: Version 6.0.1, compiled on Apr 15 2015 at 19:59:58
             Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
             Copyright (c) 2007-2009 Joerg Wunsch

             System wide configuration file is "D:\Program Files (x86)\arduino-P
R-3640-BUILD-375\hardware\tools\avr/etc/avrdude.conf"

             Using Port                    : COM14
             Using Programmer              : avr109
             Overriding Baud Rate          : 1200
             AVR Part                      : AT90USB82
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PC6
             RESET disposition             : possible i/o
             RETRY pulse                   : SCK
             serial program mode           : yes
             parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             ByteDelay                     : 0
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                      Block Poll               Page
          Polled
               Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW
 MaxW   ReadBack
               ----------- ---- ----- ----- ---- ------ ------ ---- ------ -----
 ----- ---------
               eeprom        65    20     4    0 no        512    4    128  9000
  9000 0x00 0x00
               flash         65     6   128    0 yes      8192  128     64  4500
  4500 0x00 0x00
               lfuse          0     0     0    0 no          1    0      0  9000
  9000 0x00 0x00
               hfuse          0     0     0    0 no          1    0      0  9000
  9000 0x00 0x00
               efuse          0     0     0    0 no          1    0      0  9000
  9000 0x00 0x00
               lock           0     0     0    0 no          1    0      0  9000
  9000 0x00 0x00
               calibration    0     0     0    0 no          1    0      0     0
     0 0x00 0x00
               signature      0     0     0    0 no          3    0      0     0
     0 0x00 0x00

             Programmer Type : butterfly
             Description     : Atmel AppNote AVR109 Boot Loader

Connecting to programmer: .
Found programmer: Id = "HL2.0.5"; type = S
    Software Version = 1.0; No Hardware Version given.
Programmer supports auto addr increment.
Programmer supports buffered memory access with buffersize=128 bytes.

Programmer supports the following devices:
    Device code: 0x44

avrdude.exe: devcode selected: 0x44
avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude.exe: Device signature = 0x1e9389
avrdude.exe: Expected signature for AT90USB82 is 1E 93 82
             Double check chip, or use -F to override this check.

avrdude.exe done.  Thank you.
```

thesymbol, 2015-08-08:
notice the 
avrdude.exe: Device signature = 0x1e9389
avrdude.exe: Expected signature for AT90USB82 is 1E 93 82
             Double check chip, or use -F to override this check.

seems as if the device (bootloader maybe) is wrong for the 8u2.

NicoHood, 2015-08-08:
That is correct because the bootloader was compiled for the 8u2 so the signature shows a 8u2 device.

thesymbol, 2015-08-08:
going to wait for you to compile it for at90usb82 then.

[44 more replies]

closed 2016-06-26 by NicoHood