NicoHood/HoodLoader2 #10: json file error

pierrotm777, 2015-06-01:
your link https://github.com/NicoHood/HoodLoader2/releases/download/2.0.4/2.0.4-boards_manager.zip
return an error

NicoHood, 2015-06-01:
now online, missed the upload.
If it works, please close the issue. Thanks for the report!

pierrotm777, 2015-06-01:
Return me again an error

	... 1 more
com.fasterxml.jackson.core.JsonParseException: Unexpected character ('<' (code 60)): expected a valid value (number, String, array, object, 'true', 'false' or 'null')
 at [Source: java.io.FileInputStream@5ccbe0; line: 4, column: 2]

per1234, 2015-06-01:
I just installed it under 1.6.4 and the hourly build with no problem. Maybe the problem is caused by the first failed install attempt before NicoHood had uploaded the archive. @pierrotm777 try deleting:

arduino15/package_NicoHood_HoodLoader2_index.json
arduino15/package_NicoHood_HoodLoader2_index.json.sig.tmp
arduino15/packages/HoodLoader2/
arduino15/staging/packages/2.0.4-boards_manager.zip

and then try installing again and let us know if you still have the problem.

NicoHood, 2015-06-15:
Issue solved?

pierrotm777, 2015-06-15:
Yes , thanks :-)

closed 2015-06-15 by NicoHood