NicoHood/HoodLoader2 #14: 32u4 + 328p As a UNO with greater opportunities.

bgma, 2015-06-24:
Hello,
I wonder if it is possible to use HoodLoader  to Pololu - A-Star 32u4 Micro & 328p?
/ Grateful for answers BG

[image]

NicoHood, 2015-06-24:
Theoretically yes, but the 4kb limit is just too much for a 32u4. The USB core of the u4 is bigger than the u2 it seems. At least it wont compile within 4kb for a 32u4. You have to optimize the code even more. Feel free to do so, but I am not sure if this is possible if you do not remove the whole Lufa core stuff.

If it would be possible I'd create an own board with those two mcus combined. But right now you have to stick with the 16u2 or 32u2, which is still pretty powerful if you use it wisely.

bgma, 2015-06-24:
Seems like there is a lot to do to make it work. I do not think I have enough knowledge to solve this.

NicoHood, 2015-07-22:
The last commit is very close to compile on a 32u4.
https://github.com/NicoHood/HoodLoader2/commit/7fab825897e7e721eb76788e3f6f5bcf8505a15a

If you deactivate the lock byte support, remove the brr optimization and revert to the inlined stream function I only get 24 bytes off. I switched from RAM descriptors to PROGMEM. If we revert this back again it could save even more flash and compile fine.

NicoHood, 2015-07-25:
Should work with the current dev version (fast-usbserial branch).
Not tested, but compiles. Reset pin is PD4 (Digital pin 4).

Feel free to test it.

bgma, 2015-07-30:
Hello Nico,

I get it not to work.
Get the following results.
I have tested with Pololu-a-star-32u4 and ArduinoLeonardo but get the same results.
Maybe I am blind, or just do not see what I'm doing wrong?
Reset pin is PD4 (Digital pin 4).????

```
Attempting to enter programming mode ...
Entered programming mode OK.
Signature = 0x1E 0x95 0x87 
Processor = ATmega32U4
Flash memory size = 32768 bytes.
LFuse = 0xFF 
HFuse = 0xD8 
EFuse = 0xCB 
Lock byte = 0xEF 
Clock calibration = 0x64 
No bootloader support for this device.
Programming mode off.
Type 'C' when ready to continue with another chip ...
```
[image]

NicoHood, 2015-07-31:
You need to a) recompile the source and b) use an ISP, the installation sketch is not updated yet.

bgma, 2015-07-31:
what do I need to recompile?

NicoHood, 2015-07-31:
Read the developer information in the wiki or the not yet closed blog. The best thing you can do is use a normal linux machine to recompile with avr gcc 5.1

Or you wait some days till I compile it for you

bgma, 2015-07-31:
It would be very kind of you!
 I am happy to wait!

NicoHood, 2015-08-04:
Some more Feedback from me:
I got a very first setup working!

The code a) now compiles fine for 32u4 mostly all its new features as well (2M full duplex!) and b) can upload code to the 32u4 and 328 as well.

My setup was like this:
Pro micro with HoodLoader 2.0.5 and an Arduino Nano with Arduino Uno Bootloader (to get 115200 baud, 57600 currently has problems). 5v and gnd connected, TX + RX crossed. **Important** put a 100nF capacitor between Digital Pin 4 of the leonardo/micro and the 328 reset line. Otherwise it wont work.

I am currently resolving the 57600 baud rate issue for nanos/pro minis. Then I will compile all bootloader hex files and finally Update the Installation sketch once its all done.

If you have an ISP (you can use any arduino as ISP, see the wiki) you can upload the new bootloader in a few hours maybe. If the compiled hex version is in the fast-usbserial branch you can upload it yourself. I will not be able to update the installation sketch until I manage to get a Windows installation. The converting script was written for windows which I dont have anymore.

NicoHood, 2015-08-05:
https://github.com/NicoHood/HoodLoader2/wiki/Burning-the-bootloader-via-ISP-%28advanced%29
https://github.com/NicoHood/HoodLoader2/wiki/HoodLoader2-32u4-Usage

I used your picture, is that okay for you?
Upload the ISP sketch (leonardo fix is in the HID Project, the normal ISP sketch will only work for an Uno) and flash the new firmware as described.

The hex files are not all tested but should work. If you can test some of them feel free to do. I also need someone to convert those .hex files into a c array which is readable by the installation sketch. Instructions here: https://github.com/NicoHood/HoodLoader2/wiki/Developer-Information#changes-to-the-atmega-bootloader-programmer

bgma, 2015-08-05:
Yes it is ok.
Glad to be able to help those who help.
I'll test (Earliest Saturday) ... and to convert ...
/ Sincerely BG
 
Date: Wed, 5 Aug 2015 11:35:00 -0700

https://github.com/NicoHood/HoodLoader2/wiki/Burning-the-bootloader-via-ISP-%28advanced%29

https://github.com/NicoHood/HoodLoader2/wiki/HoodLoader2-32u4-Usage

NicoHood, 2015-08-06:
Made the program work under linux with wine.
Installation sketch can be tested now:
https://github.com/NicoHood/HoodLoader2/tree/fast-usbserial

Please let me know if a) installation sketch works and b) HL2 for any MCU you can test. 32u4 might be interesting but 16u2 should also be tested from someone else.

I am currently updating the wiki with the new information. Should be enough that you can use a 32u4 now I hope. Any improvements are appreciated.

NicoHood, 2015-08-13:
Any tests from your side?

NicoHood, 2015-09-26:
Closing this now, as it is now implemented and seems to work properly.
Nano support got removed, you better burn a proper, faster uno bootloader to the 32u4.
Leonardo support was more important here.

closed 2015-09-26 by NicoHood

NicoHood, 2019-02-12:
For this usecase you dont really need this bootloader. But the installation is described in the wiki.

geyseran, 2019-02-12:
If you say that I do not need the bootloader, then could you by any chance assist with a sample sketch that will send control modifier "F2" every 5 seconds for a period of 30 seconds while a computer is booting up. I would like to see if this will start the computer bios? I must apologize, I am still new in this field?

NicoHood, 2019-02-13:
Checkout this:
https://github.com/NicoHood/HID/blob/master/examples/Keyboard/BootKeyboard/BootKeyboard.ino

geyseran, 2019-02-13:
I have loaded the BootKeyboard.ino into a Pololu 32u4 micro. It works fine once the pc has booted. If I enter the pc bios, then the keys are not detected at all? I need it to work within bios level? Is this sketch specific to a certain Arduino models or can any 32u4 unit be used?

NicoHood, 2019-02-17:
The library works on all supported platform and the 32u4 is the most common. Its a known problem, that it does not work on every bios, so bad luck for you.