NicoHood/HoodLoader2 #16: Arduino Due support

NicoHood, 2015-07-25:
If desired I could add this feature, but I currently do not own the hardware.

cimd, 2015-08-21:
Hi, on the wiki pages you mentioned that some modifications on the code are needed to make it work for the Due. Can you explain what kind of modifications are needed? I`d like to use the HoodLoader2 with my Due. Thanks.

NicoHood, 2015-08-22:
Hi,
yes it is possible to make that work with due. It could be a bit complicated because of the very few space we have left, but with a 16u2 it should be possible, 32u2 even better (normally not on a  due though).

You need to:
edit the appconfig.h file and disable the Lock fuse setting (to get more free flash).
And you need to change the reset behavior. The Due needs to be erased before its reflashed. There is a new pin for that and the reset pin might be on a different pin. You need to unsure the delay is correct, otherwise it wont work (helped some with a due bootloader some time ago). so simply Triggering the lines without the delay wont work. Look at the schematic for the pins.

You can have a look at the Arduino due bootloader within the Arduino project. Must be somewhere there. The biggest challenge would be to fit it into the 4kb, because you need to implement this delete + reset with delay thing.

If you have something working feel free to post it here, also as PR.

I could also have a look at this, but I dont own a due and wont buy it just for adding due support and then not using it probably. Its not that cheap. If you buy me one I could do it though. Or you ship me one for testing purposes and I ship it back (3,5€ shipping world wide). I cant guarantee to fit it into the 4kb limit but I can try.

But cant you use the dues native USB functions? I know a 2nd MCU is always cool so that HL2 can give you more options and also an AVR instead of ARM on one board.

cimd, 2015-08-23:
Hi, Nico.

Thanks for the prompt reply. I'll take a look at the modifications you mentioned. Although the 4Kb may be indeed a challenge. I wanted to make use of the MIDI-USB features for the 16U2. I can probably do the same for the SAM3X8E, but it will also require modification of the  current MIDI library as it does not support the Due. 

How did you compile your code to the 16U2? Could't find board description for it.

NicoHood, 2015-08-23:
As said:
https://github.com/NicoHood/HoodLoader2/wiki/Developer-Information

cimd, 2015-08-23:
The links redirects to your wordpress blog, which is closed. :-)

NicoHood, 2015-08-23:
Correct. Docs moved here:
https://github.com/NicoHood/NicoHood.github.io/wiki/Installing-avr-gcc-4.8.1-and-Arduino-IDE-1.6-on-Raspberry-Pi

All you need to do is basically install avr-gcc avrlibc and use make to compile. avr-gcc of course from the link in the wiki. 5.1 is required.

NicoHood, 2015-08-26:
Any progress from you? Have you got it compiling?

cimd, 2015-08-26:
Hi, Nico. Didn't try it yet. I can only work on that during the weekends.
Thanks for the follow up.

Rgds,

Christian

NicoHood, 2015-08-27:
I was trying to add Due support myself right now but I noticed the due uses baud 1200 for reprogramming the Due. But HL2.0.5 uses 1200 for 16u2 reprogramming.
https://github.com/arduino/Arduino/blob/master/hardware/arduino/sam/firmwares/atmega16u2/arduino-usbserial/Arduino-usbserial.c#L301

This is a problem. We could create a new boards.txt entry with a different baud rate but that would mess up the whole thing.

The problem here is:
normally the CDC reprogramming uses baudrate 57600. But this is used for the Arduino Nano, which we cant use for reprogramming. Not all baud rates work with avrdude right now (there is an opened issue at the Arduino project). Thatswhy I choose 1200. But this now conflicts with the due.

We could: use 57600 here instead because this 16u2 will never see an Arduino Nano
Use a different baud rate and block that for reprogramming, which is kinda bad since we then reserv 1200, 57600 and another baud.

I tend to use 57600 for u2 devices and 1200 for u4 devices. If you can follow...

cimd, 2015-08-28:
What I missed is: so what is the downside of using 57600 baud rate? Doesn't
AVRdude work with that?

PS: have you seen this: http://radikalbytes.com/?p=17? Haven't tried it yet.

NicoHood, 2015-08-28:
HL2.0.5 now support the leonardo and micro. Meaning you can connect a Micro with a Nano for example and have a very powerful double microcontroller setup.
https://github.com/NicoHood/HoodLoader2/wiki/HoodLoader2-32u4-Usage

Baud 57600 is used to reprogram the Nano. Thatswhy we cannot use this to reprogram the 32u4 itself. For u2 devices this doesnt matter too much, since they will probably never see a Nano bootloader.

There are 3 solutions:
* Force users to also update the Nano BootLoader (a uno bootloader will also work and is faster/smaller).
* Use a baud that works for all (but most are reserved)
* use 1200 for u4 and 57600 for u2 devices.

Last option would be the best I think...
Still a problem though.

NicoHood, 2015-08-28:
The Link you posted will work with this. and you can switch between hoodloader and midi very quick.
You have to program the midi - Serial bridge yourself, but there is a USB Midi library for Arduino already. I should add an example for this maybe.

NicoHood, 2015-08-28:
I need to know at which speed the DUE is reprogrammed. It uses 1200 as baud rate for reseting the due but the speed is so slow it cannot be the final speed. Can you find out at what speed this is running? Maybe from the verbose output

https://github.com/arduino/Arduino/blob/master/hardware/arduino/sam/boards.txt#L25-L46

cimd, 2015-08-28:
Sure. I can do that once I get home tonight.

NicoHood, 2015-08-28:
Another problem:
We use the 1200baud touch to reset a 16u2 sketch and reprogram the 16u2.
If the bootloader is already running this is expected to do nothing and then reprogram the 16u2.
But for the due this will erase the due chip as well for every 16u2 reprogramming.

Not sure how to overcome this problem.

NicoHood, 2015-08-29:
Its turns out to be a serious problem. We'd need a different Serial touch baud rate which is not implemented in the IDE.

The only solution would be to detect if the IDE sets the due programming speed and erase the due then on this baud rate. We'd ignore the 1200 touch then. Not sure if the timing then works. Also setting the baud to this speed always causes a chip erase (as for the speed 1200 right now).

I dont have a due, so tell me: Is the 16u2 USB port used often? Also for debugging like Serial.print("debug msg");? If so the programming baud rate is important here. If that is something common which I except, there is no real solution I can imagine of how to fix this. The IDE needs to be changed for this and it takes forever for them to accept a patch, even more for this personal use only. You could recompile the IDE yourself but I am not sure if thats a real solution for long time use.

I read that you can also program the DUE via its 2nd USB port. Does this work reliable? If yes we could avoid using the 16u2 as DUE reprogrammer and only make use of the 16u2 self reprogramming.

For example you could just use the HL2 as CDC bootloader for the 16u2 and then switch betrween reprogramming firmware if really needed (if you messed up the dues program, so it cannot reprogram itself anymore with the normal usb port). This limits the features somehow more. Its not a full hoodLoader, it would be more a CDC bootloader only where you have to upload the DUE uploading firmware. (similar to the uno, just with CDC instead of DFU and that the DUE still has a 2nd USB port that can be used).

If you can follow...

cimd, 2015-08-29:
I actually prefer using the 16u2 USB port then the native USB port, as
every once and a while the native port gets disconnected when uploading
sketches. Well, its not really an issue, as all it takes is disconnecting
and reconnecting the board. But having a second programmable
micro-controller would definitely be a plus. Well, long story short, yes
the native USB port can be used for programming, and the 16u2 port could be
used just for programming itself. And case needed, onecould always flash
back the original 16u2 onto it.

cimd, 2015-08-29:
Do you still need to know at what speed the Due is reprogrammed?

NicoHood, 2015-08-30:
Yes that information would help. Maybe I can imagine of a different Idea. But I guess I need to get a Due myself to get this working reliable then. Turns out not to be so easy.

cimd, 2015-08-30:
It's very frustrating that most libraries just won't work with the Due.
Right now I was trying to use the Bounce2 library for the same project I
needed the HL2. And it turns out that it doesn't work with the Due either.
I'll have to create my own classes. Using Due has its benefits, but it is
very disappointing most of the times.

NicoHood, 2015-08-30:
The Teensy might be a better choice. It also has very good USB features.

If you know the dues programming baud that'd still be nice. But I think its not worth to put more work into that because there are too many problems not solved yet and too hard to fix.

NicoHood, 2015-09-24:
I tried again to fix this 57600 baud problem with the nano and ended up using 57600 again (for better Leonardo compatibility). There is no way to differentiate between a 16u2 boot process and nano/due process. More work has to be done to add this.

I will close this issue as "Impossible to add right now"

cimd, 2015-09-24:
Thanks for the feedback, Nico. But it also looks like I may need to move on
to something with more horsepower then the Due for my application:
Electronic drum module with sounds . I already bought a STM32F4 Discovery
board, due to its DSP capabilities, for testing. Let's see.

NicoHood, 2015-09-24:
Alright, thanks for you feedback.

You might want to use a Teensy 3.2. I have no idea what DSP is but it seems you can use it somehow:
https://forum.pjrc.com/threads/23682-Teensy-3-0-DSP-libraries

NicoHood, 2015-09-26:
Reopening since a jumper could really solve this issue. You then have to explicit select the mode and some things would work different but this could be a solution to get it working.

Similar Idea for autoreset:
https://github.com/PeterVH/HoodLoader2/commit/fdff871d83c20b1e3201c7694607d88bf90d673b

However I'd need a Due to test this.

xcvista, 2016-06-25:
I think that there isn't too much to worry about the bootloader in Due. It is OTP programmed in the chip in factory (mask ROM?) triggered by either jumping into it from application code (implemented as part of Due's default USB stack, not the bootloader) or when the main ROM is empty (that is, just after a memory wipe,) and use an alternative programming tool.

NicoHood, 2016-06-26:
You are right. There is less sense in using the 16u2 with the due, as it is powerfull enough and already has a USB-Core. Snooping the protocol is something I did befor with hoodloader1 (in a different way) and I am not going to do this again :D

closed 2016-06-26 by NicoHood

xcvista, 2016-06-26:
In fact I would like to see a version of HoodLoader 16U2 that does NOT support programming the ARM for the Due. The point here is that the reprogrammed 16U2 would be used for something other than a USB serial adapter.

For the ARM on the Due there really isn't any need for a bootloader. The USB+Serial bootloader is mask ROMed into the chip and the hardware erase is implemented not only as a signal from the 16U2 but also a physical button, so even including wiping the USB stack or the 16U2 won't brick it at all. This can result in something interesting applications for the Due-HoodLoader combo, as the USB stack on the ARM is really an 480Mbps high-speed OTG port (capable of being both device and host). For example a network-capable KVM switch.

NicoHood, 2016-06-26:
Just install the CDC LUFA bootloader then. I made a PR for arduino compatiblity.

xcvista, 2016-06-26:
How to get that into my Arduino IDE?

NicoHood, 2016-06-26:
Just burn the bootloader and select HoodLoader2 16u2 for compiling firmwares.

NicoHood, 2020-08-23:
From your lsusb command we can see that there is stil a DFU bootloader on the device.

What is DFU? -> DFU is an atmel bootloader (or protocol) that is tl;dr not arduino IDE compatible. The arduino CDC bootloader is. Hoodloader 2 is a CDC bootloader (to program itself) AND a usb-serial bridge (to program the IO MCU).

Now back to topic:
1. You need to burn the bootloader
2. If the file is not there, please find out why. It should have shipped with the board manager. Did the arduino team change any file path? I dont know, but I think you will manage! Please report.

wingedrhino, 2020-08-24:
> From your lsusb command we can see that there is stil a DFU bootloader on the device.

> What is DFU? -> DFU is an atmel bootloader (or protocol) that is tl;dr not arduino IDE compatible. The arduino CDC bootloader is. Hoodloader 2 is a CDC bootloader (to program itself) AND a usb-serial bridge (to program the IO MCU).

Aah okay. So I installed the wrong bootloader.

> Now back to topic:
> 
>     1. You need to burn the bootloader
> 
>     2. If the file is not there, please find out why. It should have shipped with the board manager. Did the arduino team change any file path? I dont know, but I think you will manage! Please report.

I'm on Manjaro, Arduino IDE 1.8.13. HoodLoader2 is installed at `~/.arduino15/packages/HoodLoader2/hardware/avr/2.0.5/`.

Here is the `boards.txt` file within: https://github.com/NicoHood/HoodLoader2/blob/master/avr/boards.txt#L53

At the line I linked to, it seems you expect there to be a `~/.arduino15/packages/HoodLoader2/hardware/avr/2.0.5/bootloaders/HoodLoader2/HoodLoader2.hex` file, which doesn't exist here: https://github.com/NicoHood/HoodLoader2/tree/master/avr/bootloaders/HoodLoader2. It also doesn't show up after running the builds via the makefile in that directory.

Which is the right board to select to burn a bootloader to a "standalone" 16u2?

I see that the hexfiles for the [Uno16u2](https://raw.githubusercontent.com/NicoHood/HoodLoader2/master/avr/bootloaders/HexFiles/HoodLoader2_0_5_Uno_atmega16u2.hex) and [Mega16u2](https://raw.githubusercontent.com/NicoHood/HoodLoader2/master/avr/bootloaders/HexFiles/HoodLoader2_0_5_Mega_atmega16u2.hex) are both different. What do these do differently on the inside? Because I'd assume I should be able to burn either of these to the Due straight and it'd work?

NicoHood, 2020-08-24:
uno and mega have different usb PIDs to identify for the windows driver. And they have a different name shown in lsusb. The rest is the same.

Did you install hoodloader2 via master branch or via board manager? board manager should work, master... i am not sure.

As a quickfix, just create/move into the right folder. select the correct board and hit 'burn bootloader'.
BTW: You can also use the arduino install sketch, if you follow the readme in the wiki.... that is way simpler I guess.

wingedrhino, 2020-08-24:
> uno and mega have different usb PIDs to identify for the windows driver. And they have a different name shown in lsusb. The rest is the same.

So I did NOT know that. This means only the name changes. I installed the HoodLoader2 Mega bootloader instead lol. It works. Was able to then upload sketch via USB.

> Did you install hoodloader2 via master branch or via board manager? board manager should work, master... i am not sure.

This file is missing in both `master` and `v2.0.5`. Here is the `v2.0.5` branch: https://github.com/NicoHood/HoodLoader2/tree/2.0.5/avr/bootloaders/HexFiles

> As a quickfix, just create/move into the right folder. select the correct board and hit 'burn bootloader'.

Like I said earlier, I just installed the Mega Bootloader.

> BTW: You can also use the arduino install sketch, if you follow the readme in the wiki.... that is way simpler I guess.

By now I've gotten used to the ISP connections lol.