NicoHood/HoodLoader2 #27: How to restore stock 16u2 bootloader?

grasmanek94, 2016-03-10:
I want to restore the stock 16u2 bootloader on my Arduino Uno R3 using another Arduino Uno R3. Which hex file do I need to select?

NicoHood, 2016-03-10:
This one:
https://github.com/NicoHood/HoodLoader2/blob/master/avr/boards.txt#L56

But normally this is never needed, since HoodLoader2 has exactly the same features, its just better.
Also see: https://github.com/NicoHood/HoodLoader2/blob/master/avr/examples/RunBootloader/RunBootloader.ino

Next time contact me via contact.nicohood.de and not as a github issue. I will not further respond here. Thanks for your understanding.

I will only leave this open, until I add this information to the wiki FAQ page.

grasmanek94, 2016-03-10:
Ah alright, thank you!

NicoHood, 2016-04-16:
Actually it was always there. I just renamed the heading now to make it more clear and added a link:
https://github.com/NicoHood/HoodLoader2/wiki/Troubleshoot-FAQ#restore-stock-16u2-bootloader

closed 2016-04-16 by NicoHood