NicoHood/HoodLoader2 #28: MAGIC_KEY not defined when building for 16u2

tknapstad, 2016-05-13:
After a recent Arduino IDE update, code for 16u2 fails to compile because MAGIC_KEY is not defined.

```shell
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os  -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega16u2 -DF_CPU=16000000L -DARDUINO=10609 -DARDUINO_HOODLOADER2 -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x484C '-DUSB_MANUFACTURER="Nico Hood"' '-DUSB_PRODUCT="HoodLoader2 16u2"' -DMAGIC_KEY_POS=(RAMEND-1) -DUSB_EP_SIZE=16 "-IC:\Users\Tarjei\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.11\cores\arduino" "-IC:\Users\Tarjei\AppData\Local\Arduino15\packages\HoodLoader2\hardware\avr\2.0.5\variants\HoodLoader2" "C:\Users\Tarjei\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.11\cores\arduino\CDC.cpp" -o "C:\Users\Tarjei\AppData\Local\Temp\buildf6aa37d3ca1ba04de8acf1acc6efd411.tmp\core\CDC.cpp.o"
C:\Users\Tarjei\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.11\cores\arduino\CDC.cpp: In function 'bool CDC_Setup(USBSetup&)':

C:\Users\Tarjei\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.11\cores\arduino\CDC.cpp:129:34: error: 'MAGIC_KEY' was not declared in this scope

     *(uint16_t *)magic_key_pos = MAGIC_KEY; 
```

It's defined in variants/leonardo/pins_arduino.h, but this doesn't seem to get included where it needs to. 

```shell
grep -r "MAGIC_KEY" *
variants/leonardo/pins_arduino.h:#ifndef MAGIC_KEY
variants/leonardo/pins_arduino.h:#define MAGIC_KEY 0x7777
```

Modifying boards.txt fixes the build, but I'm not sure if this is the correct magic key?

```
HoodLoader2atmega16u2.build.extra_flags={build.usb_flags} -DMAGIC_KEY=0x7777 -DMAGIC_KEY_POS=(RAMEND-1) -DUSB_EP_SIZE=16
```

NicoHood, 2016-05-18:
Thx for the report. May you try this fix: https://github.com/arduino/Arduino/pull/4967

tknapstad, 2016-05-18:
Confirmed working with https://github.com/arduino/Arduino/pull/4967

The boards.txt modification should suffice until 1.6.10 is released.

Przemo2, 2016-05-28:
Can you add Magic Key definition to code? Because I don't know how to do that.

NicoHood, 2016-05-28:
Just use the nighlty IDE if it is fixed there.

Przemo2, 2016-05-28:
Do you mean this [arduino/Arduino#4967](https://github.com/arduino/Arduino/pull/4967)? I can't download it link is broken.

Przemo2, 2016-05-28:
Site is opening but all links to download files aren't working there is error 404.

Przemo2, 2016-05-31:
So can You help please?

NicoHood, 2016-05-31:
arduino.cc/download

Przemo2, 2016-06-01:
Thanks for help
I downloaded hourly build and it is working 😊

NicoHood, 2016-06-26:
Closing as the nightly IDE solves the issue. Hopefully it will come out soon.

closed 2016-06-26 by NicoHood