NicoHood/HoodLoader2 #29: Unable to upload sketches from Arduino IDE with HoodLoader2 16u2

FliiFe, 2016-05-23:
Hey.
When I choose `HoodLoader2 16u2 HoodLoader2 Uno` ad board, and press upload, the IDE "fakes" an upload, i.e it uploads as usual, showing the `Found programmer: Id = "HL2.0.5"; type = S` message, the arduino RX and TX blink for like a tenth of a second and then the old program runs.

I have an Arduino Uno (from `arduino.org`, not `.cc`. Does that matter ?) with HoodLoader2.

Output of `udevadm monitor --subsystem-match=usb --property | grep -E "REVISION|VENDOR|MODEL"` : 
```
ID_VENDOR_FROM_DATABASE=Arduino SA
ID_VENDOR_FROM_DATABASE=Arduino SA
ID_MODEL=HoodLoader2_16u2
ID_MODEL_ENC=HoodLoader2\x2016u2
ID_MODEL_ID=484c
ID_REVISION=0100
ID_VENDOR=Arduino_LLC
ID_VENDOR_ENC=Arduino\x20LLC
ID_VENDOR_FROM_DATABASE=Arduino SA
ID_VENDOR_ID=2341
ID_MODEL=HoodLoader2_16u2
ID_MODEL_ENC=HoodLoader2\x2016u2
ID_MODEL_ID=484c
ID_REVISION=0100
ID_VENDOR=Arduino_LLC
ID_VENDOR_ENC=Arduino\x20LLC
ID_VENDOR_FROM_DATABASE=Arduino SA
ID_VENDOR_ID=2341
```

Output of `lsusb -d 2341: `: 
```
Bus 003 Device 068: ID 2341:484c Arduino SA 
```

Am I doing something wrong, or is this an issue ?

NicoHood, 2016-05-24:
It seems that the correct bootloader is installed. The manufacturer does not matter, as I use the .cc VID.

Which Arduino IDE do you use? The one from .org? I never tested the .org IDE. Please try the latest nightly build of Arduino IDE 1.6.10. If it still does not work, please post the full debug log of the ide.

FliiFe, 2016-05-24:
I am using the latest 1.6.10 nightly from .cc, because the 1.6.9 would not compile (#28).
Here is the full compilation+upload log : 
```
/home/fliife/arduino-1.6.10/arduino-builder -dump-prefs -logger=machine -hardware "/home/fliife/arduino-1.6.10/hardware" -hardware "/home/fliife/.arduino15/packages" -tools "/home/fliife/arduino-1.6.10/tools-builder" -tools "/home/fliife/arduino-1.6.10/hardware/tools/avr" -tools "/home/fliife/.arduino15/packages" -built-in-libraries "/home/fliife/arduino-1.6.10/libraries" -libraries "/home/fliife/Arduino/libraries" -fqbn=HoodLoader2:avr:HoodLoader2atmega16u2:board=uno -vid-pid=0X2341_0X0043 -ide-version=10610 -build-path "/tmp/buildbeb4636d72090460f459d638108b4329.tmp" -warnings=all -prefs=build.warn_data_percentage=75 -verbose "/tmp/arduino_modified_sketch_932987/Blink.ino"
/home/fliife/arduino-1.6.10/arduino-builder -compile -logger=machine -hardware "/home/fliife/arduino-1.6.10/hardware" -hardware "/home/fliife/.arduino15/packages" -tools "/home/fliife/arduino-1.6.10/tools-builder" -tools "/home/fliife/arduino-1.6.10/hardware/tools/avr" -tools "/home/fliife/.arduino15/packages" -built-in-libraries "/home/fliife/arduino-1.6.10/libraries" -libraries "/home/fliife/Arduino/libraries" -fqbn=HoodLoader2:avr:HoodLoader2atmega16u2:board=uno -vid-pid=0X2341_0X0043 -ide-version=10610 -build-path "/tmp/buildbeb4636d72090460f459d638108b4329.tmp" -warnings=all -prefs=build.warn_data_percentage=75 -verbose "/tmp/arduino_modified_sketch_932987/Blink.ino"
Build options changed, rebuilding all
"/home/fliife/arduino-1.6.10/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega16u2 -DF_CPU=16000000L -DARDUINO=10610 -DARDUINO_HOODLOADER2 -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x484C '-DUSB_MANUFACTURER="Nico Hood"' '-DUSB_PRODUCT="HoodLoader2 16u2"' -DMAGIC_KEY_POS=(RAMEND-1) -DUSB_EP_SIZE=16 "-I/home/fliife/arduino-1.6.10/hardware/arduino/avr/cores/arduino" "-I/home/fliife/.arduino15/packages/HoodLoader2/hardware/avr/2.0.5/variants/HoodLoader2" "/tmp/buildbeb4636d72090460f459d638108b4329.tmp/sketch/Blink.ino.cpp" -o "/dev/null"
"/home/fliife/arduino-1.6.10/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega16u2 -DF_CPU=16000000L -DARDUINO=10610 -DARDUINO_HOODLOADER2 -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x484C '-DUSB_MANUFACTURER="Nico Hood"' '-DUSB_PRODUCT="HoodLoader2 16u2"' -DMAGIC_KEY_POS=(RAMEND-1) -DUSB_EP_SIZE=16 "-I/home/fliife/arduino-1.6.10/hardware/arduino/avr/cores/arduino" "-I/home/fliife/.arduino15/packages/HoodLoader2/hardware/avr/2.0.5/variants/HoodLoader2" "/tmp/buildbeb4636d72090460f459d638108b4329.tmp/sketch/Blink.ino.cpp" -o "/dev/null"
"/home/fliife/arduino-1.6.10/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega16u2 -DF_CPU=16000000L -DARDUINO=10610 -DARDUINO_HOODLOADER2 -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x484C '-DUSB_MANUFACTURER="Nico Hood"' '-DUSB_PRODUCT="HoodLoader2 16u2"' -DMAGIC_KEY_POS=(RAMEND-1) -DUSB_EP_SIZE=16 "-I/home/fliife/arduino-1.6.10/hardware/arduino/avr/cores/arduino" "-I/home/fliife/.arduino15/packages/HoodLoader2/hardware/avr/2.0.5/variants/HoodLoader2" "/tmp/buildbeb4636d72090460f459d638108b4329.tmp/sketch/Blink.ino.cpp" -o "/tmp/buildbeb4636d72090460f459d638108b4329.tmp/preproc/ctags_target_for_gcc_minus_e.cpp"
"/home/fliife/arduino-1.6.10/tools-builder/ctags/5.8-arduino10/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "/tmp/buildbeb4636d72090460f459d638108b4329.tmp/preproc/ctags_target_for_gcc_minus_e.cpp"
"/home/fliife/arduino-1.6.10/hardware/tools/avr/bin/avr-g++" -c -g -Os -Wall -Wextra -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega16u2 -DF_CPU=16000000L -DARDUINO=10610 -DARDUINO_HOODLOADER2 -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x484C '-DUSB_MANUFACTURER="Nico Hood"' '-DUSB_PRODUCT="HoodLoader2 16u2"' -DMAGIC_KEY_POS=(RAMEND-1) -DUSB_EP_SIZE=16 "-I/home/fliife/arduino-1.6.10/hardware/arduino/avr/cores/arduino" "-I/home/fliife/.arduino15/packages/HoodLoader2/hardware/avr/2.0.5/variants/HoodLoader2" "/tmp/buildbeb4636d72090460f459d638108b4329.tmp/sketch/Blink.ino.cpp" -o "/tmp/buildbeb4636d72090460f459d638108b4329.tmp/sketch/Blink.ino.cpp.o"
"/home/fliife/arduino-1.6.10/hardware/tools/avr/bin/avr-gcc" -c -g -x assembler-with-cpp -mmcu=atmega16u2 -DF_CPU=16000000L -DARDUINO=10610 -DARDUINO_HOODLOADER2 -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x484C '-DUSB_MANUFACTURER="Nico Hood"' '-DUSB_PRODUCT="HoodLoader2 16u2"' -DMAGIC_KEY_POS=(RAMEND-1) -DUSB_EP_SIZE=16 "-I/home/fliife/arduino-1.6.10/hardware/arduino/avr/cores/arduino" "-I/home/fliife/.arduino15/packages/HoodLoader2/hardware/avr/2.0.5/variants/HoodLoader2" "/home/fliife/arduino-1.6.10/hardware/arduino/avr/cores/arduino/wiring_pulse.S" -o "/tmp/buildbeb4636d72090460f459d638108b4329.tmp/core/wiring_pulse.S.o"
"/home/fliife/arduino-1.6.10/hardware/tools/avr/bin/avr-gcc" -c -g -Os -Wall -Wextra -std=gnu11 -ffunction-sections -fdata-sections -MMD -mmcu=atmega16u2 -DF_CPU=16000000L -DARDUINO=10610 -DARDUINO_HOODLOADER2 -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x484C '-DUSB_MANUFACTURER="Nico Hood"' '-DU [... cut ...]
avrdude: verifying flash memory against /tmp/buildbeb4636d72090460f459d638108b4329.tmp/Blink.ino.hex:
avrdude: load data flash data from input file /tmp/buildbeb4636d72090460f459d638108b4329.tmp/Blink.ino.hex:
avrdude: input file /tmp/buildbeb4636d72090460f459d638108b4329.tmp/Blink.ino.hex contains 4218 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.02s

avrdude: verifying ...
avrdude: 4218 bytes of flash verified

avrdude done.  Thank you.

```

NicoHood, 2016-05-24:
And you are sure the sketch does not change? You know that you are compiling a 16u2 firmware and not a 328 firmware? So the onboard led will never ever flash. That is the concept of hoodloader2. You have 2 mcus now, not only a single one. Look at the main header picture. In the wiki is also more info about this. I think you missunderstood this and the sketch uploading actually works fine.

FliiFe, 2016-05-25:
I probably misunderstood... How can I know if the sketch works ? For example, if I try HID-Project/ImprovedKeyboard, on linux, how do I know if it works ?

NicoHood, 2016-05-25:
You can test it? there are a few IO pins and 2 available leds. See the pinlayout page of the wiki.

closed 2016-05-25 by NicoHood

FliiFe, 2016-05-25:
I've tested a slightly modified version of ImprovedKeyboard, and it works !
However, how can I control the TX and RX leds ?

NicoHood, 2016-05-26:
via digitalWrite(). See the wiki for pinlayouts please.