NicoHood/HoodLoader2 #33: Esplora?

xcvista, 2016-07-17:
I actually have an Arduino Esplora. I know that putting the Leonardo loader into it won't really affect it but what about an Esplora-specific version that still have the correct VID/PID?

NicoHood, 2016-07-17:
You are free to recompile the source yourself. I will not recompile tons of firmwares, sorry. And having it look like a leonardo is not a real problem, as this is only in bootloader mode. If the sketch runs, you will see the device you set in the IDe (since the sketch uses its own usb core).

xcvista, 2016-07-17:
Is there any pointers on how to do that? 

Or maybe make it easier by allowing setting build-time variables for VID and PID, which means a build script can quickly munch through a list and build a bunch of HoodLoader variants. Something like this:

```sh
#!/bin/bash
cat vidpid.txt | while read vid pid rest; do
    make clean
    make VID=$vid PID=$pid hoodloader.ihex
    mv hoodloader.ihex hoodloader_$vid_$pid.ihex
done
```

NicoHood, 2016-07-17:
And you'd also need a name. This is a good idea. I do not know that much about makefiles, but if you have a script like this I'd merge it and possibly also recompile some more firmwares as you wish. I compiled all of them manually. I really like the idea!

xcvista, 2016-07-18:
So is there any pointers on how to modify your code into that flexible form? (hint: use `-DVID=$(VID) -DPID=$(PID)` in Makefiles to make the VID/PID seen as global `#define`'s)

NicoHood, 2016-07-18:
Just use those definitions:
https://github.com/NicoHood/HoodLoader2/blob/master/avr/bootloaders/HoodLoader2/makefile#L44

The names are found inside the code (for leonardo etc) but you can pass real values in the makefile as far as i remember. The rest is a matter of makefile knowledge. Maybe you need to modify the makewile with a `?=` so you can overwrite the values.