NicoHood/HoodLoader2 #35: warning: deprecated conversion in installation sketch

jango-fx, 2016-09-23:
fyi: i'm getting a lot of warnings uploading the installation sketch to an UNO
```
/HoodLoader2-master/avr/examples/Installation_Sketch/Installation_Sketch.ino:462:1: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
 };  // end of signatures
 ^
 /HoodLoader2-master/avr/examples/Installation_Sketch/Installation_Sketch.ino:462:1: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
 /HoodLoader2-master/avr/examples/Installation_Sketch/Installation_Sketch.ino:462:1: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
 /HoodLoader2-master/avr/examples/Installation_Sketch/Installation_Sketch.ino:462:1: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
 /HoodLoader2-master/avr/examples/Installation_Sketch/Installation_Sketch.ino:462:1: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
 /HoodLoader2-master/avr/examples/Installation_Sketch/Installation_Sketch.ino:462:1: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
 /HoodLoader2-master/avr/examples/Installation_Sketch/Installation_Sketch.ino:462:1: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
 /HoodLoader2-master/avr/examples/Installation_Sketch/Installation_Sketch.ino:462:1: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
 /HoodLoader2-master/avr/examples/Installation_Sketch/Installation_Sketch.ino:462:1: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
 /HoodLoader2-master/avr/examples/Installation_Sketch/Installation_Sketch.ino:462:1: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
 /HoodLoader2-master/avr/examples/Installation_Sketch/Installation_Sketch.ino:462:1: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
 /HoodLoader2-master/avr/examples/Installation_Sketch/Installation_Sketch.ino:462:1: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
 /HoodLoader2-master/avr/examples/Installation_Sketch/Installation_Sketch.ino:462:1: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
 /HoodLoader2-master/avr/examples/Installation_Sketch/Installation_Sketch.ino:462:1: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
 /HoodLoader2-master/avr/examples/Installation_Sketch/Installation_Sketch.ino:462:1: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
 /HoodLoader2-master/avr/examples/Installation_Sketch/Installation_Sketch.ino:462:1: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
 /HoodLoader2-master/avr/examples/Installation_Sketch/Installation_Sketch.ino:462:1: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
 /HoodLoader2-master/avr/examples/Installation_Sketch/Installation_Sketch.ino:462:1: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
 /HoodLoader2-master/avr/examples/Installation_Sketch/Installation_Sketch.ino:462:1: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
 /HoodLoader2-master/avr/examples/Installation_Sketch/Installation_Sketch.ino:462:1: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
 /HoodLoader2-master/avr/examples/Installation_Sketch/Installation_Sketch.ino:462:1: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
 /HoodLoader2-master/avr/examples/Installation_Sketch/Installation_Sketch.ino:462:1: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
 /HoodLoader2-master/avr/examples/Installation_Sketch/Installation_Sketch.ino:462:1: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
 /HoodLoader2-master/avr/examples/Installation_Sketch/Installation_Sketch.ino:462:1: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
 /HoodLoader2-master/avr/examples/Installation_Sketch/Installation_Sketch.ino:462:1: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
 /HoodLoader2-master/avr/examples/Installation_Sketch/Installation_Sketch.ino:462:1: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
 /HoodLoader2-master/avr/examples/Installation_Sketch/Installation_Sketch.ino:462:1: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
 /HoodLoader2-master/avr/examples/Installation_Sketch/Installation_Sketch.ino:462:1: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
 /HoodLoader2-master/avr/examples/Installation_Sketch/Installation_Sketch.ino:462:1: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
 /HoodLoader2-master/avr/examples/Installation_Sketch/Installation_Sketch.ino:462:1: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
 /HoodLoader2-master/avr/examples/Installation_Sketch/Installation_Sketch.ino:462:1: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
 /HoodLoader2-master/avr/examples/Installation_Sketch/Installation_Sketch.ino: In function 'void showHex(byte, boolean, boolean)':
 /HoodLoader2-master/avr/examples/Installation_Sketch/Installation_Sketch.ino:511:38: warning: narrowing conversion of '((((int)(b >> 4)) & 255) | 48)' from 'int' to 'char' inside { } [-Wnarrowing]
   char buf [4] = { ((b >> 4) & 0x0F) | '0', (b & 0x0F) | '0', ' ' , 0 };
                                      ^
                                      /HoodLoader2-master/avr/examples/Installation_Sketch/Installation_Sketch.ino:511:56: warning: narrowing conversion of '((((int)b) & 15) | 48)' from 'int' to 'char' inside { } [-Wnarrowing]
   char buf [4] = { ((b >> 4) & 0x0F) | '0', (b & 0x0F) | '0', ' ' , 0 };
```

NicoHood, 2016-09-23:
Thats a good point, I am going to fix this when I got time.

NicoHood, 2017-03-01:
Fixed in dev branch:
https://github.com/NicoHood/HoodLoader2/commit/1f2fb1062336dd13b6d1d6d90cf4cd4b3efb6970

I am not sure if the (char*) conversion fixes are correct, I could not test yet.

closed 2017-03-01 by NicoHood