NicoHood/HoodLoader2 #36: Arduino Clones

pedro-javierf, 2016-09-29:
Hello, I know issues are not the best way to contact you, but anyways.
I have an arduino UNO that is not retail, and have some different pins.

It has some pins named ISCF.

What do I have to do so I can flash the hoodloader bootloader once it has been loaded as an sketch?
Thanks,

Pedro.
[image]
[image]

oqibidipo, 2016-09-30:
Your board has a CH340G USB-to-serial chip instead of an ATmega16u2 (or 8u2), so you can forget about putting HoodLoader2 on it.

> It has some pins named ISCF.

Looks more like ISCP (_should_ be ICSP) with part of the P outside the printable area – compare with AREF & POWER.

pedro-javierf, 2016-09-30:
Ok thanks.
Hoodloader won't work with that chip right?

closed 2016-10-03 by NicoHood