NicoHood/HoodLoader2 #38: 'HID' core not installed

Nathan-Melaku, 2017-01-14:
I have been trying to make an arduino keyboard. I am living in Ethiopia, and i can't get my hand on other arduinos except the one I got as a gift, which happens to be UNO. so I read the wiki thoroughly. and installed the hoodloader2 on the board. then I installed the hid library using the library manager, and the board definitions using the board manager. and i tried to compile the Improvedkeyboard example. but I got this error. 
 Selected board depends on 'HID' core (not installed).
i chose the hoodloader2 16u2 form the board. I have checked the sketchbook folder and it looks like this 

root@NateDawg:~/sketchbook/libraries# ls
HID  readme.txt
root@NateDawg:~/sketchbook/libraries# ls HID
examples  extras  header.jpg  keywords.txt  library.properties  Readme.md  src
root@NateDawg:~/sketchbook/libraries# 

can u pls help me I don't seem to understand my problem.

ps. I am using arduino 1.6.7

NicoHood, 2017-01-14:
can you please enable all debug output in the preferences and post the output?

You should upgrade to the newer IDE anyways. I cannot give support to such old versions, that is possibly the reason for this error.

Nathan-Melaku, 2017-01-16:
ok  I have upgraded my arduino to 1.8.1. now it did finish compiling. I've selected hoodloader 16u2 for the board. but when I try uploading this happend. 

Sketch uses 5530 bytes (45%) of program storage space. Maximum is 12288 bytes.
Global variables use 190 bytes (37%) of dynamic memory, leaving 322 bytes for local variables. Maximum is 512 bytes.
processing.app.debug.RunnerException
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:160)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:78)
	at processing.app.SketchController.upload(SketchController.java:709)
	at processing.app.SketchController.exportApplet(SketchController.java:682)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2190)
	at java.lang.Thread.run(Thread.java:745)
Caused by: processing.app.SerialException: Error touching serial port '/dev/ttyACM0'.
	at processing.app.Serial.touchForCDCReset(Serial.java:87)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:144)
	... 5 more
Caused by: jssc.SerialPortException: Port name - /dev/ttyACM0; Method name - openPort(); Exception type - Permission denied.
	at jssc.SerialPort.openPort(SerialPort.java:170)
	at processing.app.Serial.touchForCDCReset(Serial.java:81)
	... 6 more

Nathan-Melaku, 2017-01-16:
`/home/cybass/Documents/tool/arduino-1.8.1/arduino-builder -dump-prefs -logger=machine -hardware /home/cybass/Documents/tool/arduino-1.8.1/hardware -hardware /home/cybass/.arduino15/packages -tools /home/cybass/Documents/tool/arduino-1.8.1/tools-builder -tools /home/cybass/Documents/tool/arduino-1.8.1/hardware/tools/avr -tools /home/cybass/.arduino15/packages -built-in-libraries /home/cybass/Documents/tool/arduino-1.8.1/libraries -libraries /home/cybass/Arduino/libraries -fqbn=HoodLoader2:avr:HoodLoader2atmega16u2:board=uno -vid-pid=0X2341_0X0043 -ide-version=10801 -build-path /tmp/arduino_build_972169 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=/home/cybass/Documents/tool/arduino-1.8.1/hardware/tools/avr -prefs=runtime.tools.avr-gcc.path=/home/cybass/Documents/tool/arduino-1.8.1/hardware/tools/avr -prefs=runtime.tools.arduinoOTA.path=/home/cybass/Documents/tool/arduino-1.8.1/hardware/tools/avr -verbose /home/cybass/Arduino/libraries/HID-Project/examples/Keyboard/ImprovedKeyboard/ImprovedKeyboard.ino
/home/cybass/Documents/tool/arduino-1.8.1/arduino-builder -compile -logger=machine -hardware /home/cybass/Documents/tool/arduino-1.8.1/hardware -hardware /home/cybass/.arduino15/packages -tools /home/cybass/Documents/tool/arduino-1.8.1/tools-builder -tools /home/cybass/Documents/tool/arduino-1.8.1/hardware/tools/avr -tools /home/cybass/.arduino15/packages -built-in-libraries /home/cybass/Documents/tool/arduino-1.8.1/libraries -libraries /home/cybass/Arduino/libraries -fqbn=HoodLoader2:avr:HoodLoader2atmega16u2:board=uno -vid-pid=0X2341_0X0043 -ide-version=10801 -build-path /tmp/arduino_build_972169 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=/home/cybass/Documents/tool/arduino-1.8.1/hardware/tools/avr -prefs=runtime.tools.avr-gcc.path=/home/cybass/Documents/tool/arduino-1.8.1/hardware/tools/avr -prefs=runtime.tools.arduinoOTA.path=/home/cybass/Documents/tool/arduino-1.8.1/hardware/tools/avr -verbose /home/cybass/Arduino/libraries/HID-Project/examples/Keyboard/ImprovedKeyboard/ImprovedKeyboard.ino
Using board 'HoodLoader2atmega16u2' from platform in folder: /home/cybass/.arduino15/packages/HoodLoader2/hardware/avr/2.0.5
Using core 'arduino' from platform in folder: /home/cybass/Documents/tool/arduino-1.8.1/hardware/arduino/avr
Detecting libraries used...
"/home/cybass/Documents/tool/arduino-1.8.1/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega16u2 -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_HOODLOADER2 -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x484C '-DUSB_MANUFACTURER="Nico Hood"' '-DUSB_PRODUCT="HoodLoader2 16u2"' -DMAGIC_KEY_POS=(RAMEND-1) -DUSB_EP_SIZE=16 "-I/home/cybass/Documents/tool/arduino-1.8.1/hardware/arduino/avr/cores/arduino" "-I/home/cybass/.arduino15/packages/HoodLoader2/hardware/avr/2.0.5/variants/HoodLoader2" "/tmp/arduino_build_972169/sketch/ImprovedKeyboard.ino.cpp" -o "/dev/null"
"/home/cybass/Documents/tool/arduino-1.8.1/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega16u2 -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_HOODLOADER2 -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x484C '-DUSB_MANUFACTURER="Nico Hood"' '-DUSB_PRODUCT="HoodLoader2 16u2"' -DMAGIC_KEY_POS=(RAMEND-1) -DUSB_EP_SIZE=16 "-I/home/cybass/Documents/tool/arduino-1.8.1/hardware/arduino/avr/cores/arduino" "-I/home/cybass/.arduino15/packages/HoodLoader2/hardware/avr/2.0.5/variants/HoodLoader2" "-I/home/cybass/Arduino/libraries/HID-Project/src" "/tmp/arduino_build_972169/sketch/ImprovedKeyboard.ino.cpp" -o "/dev/null"
"/home/cybass/Documents/tool/arduino-1.8.1/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega16u2 -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_HOODLOADER2 -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x484C '-DUSB_MANUFACTURER="Nico Hood"' '-DUSB_PRODUCT="HoodLoader2 16u2"' -DMAGIC_KEY_POS=(RAMEND-1) -DUSB_EP_SIZE=16 "-I/home/cybass/Documents/tool/arduino-1.8.1/hardware/arduino/avr/cores/arduino" "-I/home/cybass/.arduino15/packages/HoodLoader2/hardware/avr/2.0.5/variants/HoodLoader2" "-I/home/cybass/Arduino/libraries/HID-Project/src" "-I/home/cybass/Documents/tool/arduino-1.8.1/hardware/arduino/avr/libraries/HID/src" "/tmp/arduino_build_972169/sketch/ImprovedKeyboard.ino.cpp" -o "/dev/null"
Using cached library dependencies for file: /home/cybass/Arduino/libraries/HID-Project/src/MultiReport/AbsoluteMouse.cpp
Using cached library dependencies for file: /home/cybass/Arduino/libraries/HID-Project/src/MultiReport/Consumer.cpp
Using cached library dependencies for file: /home/cybass/Arduino/libraries/HID-Project/src/MultiReport/Gamepad.cpp
Using cached library dependencies for file: /home/cybass/Arduino/libraries/HID-Project/src/MultiReport/ImprovedKeyboard.cpp
Using cached library dependencies for file: /home/cybass/Arduino/libraries/HID-Project/src/MultiReport/ImprovedMouse.cpp
Using cached library dependencies for file: /home/cybass/Arduino/libraries/HID-Project/src/MultiReport/NKROKeyboard.cpp
Using cached library dependencies for file: /home/cybass/Arduino/libraries/HID-Project/src/MultiReport/System.cpp
[... cut ...]
Caused by: processing.app.SerialException: Error touching serial port '/dev/ttyACM0'.
	at processing.app.Serial.touchForCDCReset(Serial.java:87)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:144)
	... 5 more
Caused by: jssc.SerialPortException: Port name - /dev/ttyACM0; Method name - openPort(); Exception type - Permission denied.
	at jssc.SerialPort.openPort(SerialPort.java:170)
	at processing.app.Serial.touchForCDCReset(Serial.java:81)
	... 6 more
`

NicoHood, 2017-01-16:
You need to grant your user permissions for the serial port. This is normally done by adding your user to a special group. Use google for that.

closed 2017-01-16 by NicoHood

FliiFe, 2017-01-25:
I am currently experiencing this, but my arduino ide has the correct groups and does not fail to open the serial device. It just refuses to compile anything because the "HID core" is not installed.
Installed Hoodloader through the board url and board manager