NicoHood/HoodLoader2 #43: PlatformIO

Spirevipp, 2017-04-25:
Is there any chance for some kind of PlatformIO support?
I noticed all your other libraries are on there, so it kind of makes sense that this is on there as well :)

NicoHood, 2017-04-26:
This is a bootloader with hardware definition files. Its not a library. Also I never uploaded anything to platform.io, only to the library manager. And Hoodloader2 is in the boards manager.

If you want support, open a PR. I dont know and use Platform.io.

Spirevipp, 2017-04-26:
Ok, but how would i create a pr for this? Dont i need a fork or another branch to add a pr?
From looking at the Platform.io guide for adding custom boards, it looks way out of my knowledge area.

http://docs.platformio.org/en/latest/platforms/creating_board.html

NicoHood, 2017-04-26:
Fork my project, add the required files and open a Pullup Request. Ask someone for help or a reference maybe. I can also not help with this one. Just install it the usual way ;)

NicoHood, 2017-07-05:
I am closing this as nobody seems interested in this nor can help.

closed 2017-07-05 by NicoHood