NicoHood/HoodLoader2 #46: After 1st time, 16u2 doesn't reset when serial monitor is opened

canadaduane, 2017-05-07:
After paring down some of my code, I realized that the 16u2 behaves differently in response to the serial monitor DTR reset than when using the serial monitor on the main chip (328p): after the initial serial monitor connection, the 16u2 won't reset in response to subsequent serial monitor connections. Is this intended behavior?

For instance:
```
void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);  delay(200);
  digitalWrite(LED_BUILTIN, HIGH); delay(200);
  digitalWrite(LED_BUILTIN, LOW);  delay(200);
  digitalWrite(LED_BUILTIN, HIGH); delay(200);
  digitalWrite(LED_BUILTIN, LOW);  delay(200);
  digitalWrite(LED_BUILTIN, HIGH);

  while(!Serial);
  Serial.begin(115200);
  Serial.println("I blinked");
}

void loop() {
}
```

I expected the above sketch to blink thrice, and send "I blinked" over serial, for each time I open the serial monitor.

NicoHood, 2017-05-07:
This is expected for USB MCUs and also happens with the leonardo.

closed 2017-05-07 by NicoHood

canadaduane, 2017-05-07:
Ok, thanks. BTW, hoodloader2 is awesome, thanks for writing it!