NicoHood/HoodLoader2 #47: .h files exsist but not found

supersjimmie, 2017-05-10:
I followed the procedures, including to add the board via the boards manager.
The HoodLoader2 Mega2560 is selected, as well as the comport.
When I load the sample from avr/examples/SerialKeyboard and only try to compile, it says:

```Build options changed, rebuilding all
In file included from C:\...\Arduino\sketch_may10a\sketch_may10a.ino:16:0:

C:\...\arduino-1.8.2\libraries\Keyboard\src/Keyboard.h:29:2: warning: #warning "Using legacy HID core (non pluggable)" [-Wcpp]

 #warning "Using legacy HID core (non pluggable)"

  ^

C:\...\Arduino\sketch_may10a\sketch_may10a.ino: In function 'void setup()':

sketch_may10a:28: error: 'Keyboard' not found. Does your sketch include the line '#include <Keyboard.h>'?
   Keyboard.begin();

   ^

C:\...\Arduino\sketch_may10a\sketch_may10a.ino: In function 'void loop()':

sketch_may10a:53: error: 'Keyboard' not found. Does your sketch include the line '#include <Keyboard.h>'?
     Keyboard.print(F("This message was sent with my Arduino."));

     ^

exit status 1
'Keyboard' not found. Does your sketch include the line '#include <Keyboard.h>'?
```

There is a Keyboard.h file under my libraries folder.

supersjimmie, 2017-05-10:
(Do I need to select the Hoodloader2 Mega2560 or the Hoodloader2 16u2 here?)

NicoHood, 2017-05-10:
Select Hoodloader2 16u2. You might need to update your ide and its libraries.

supersjimmie, 2017-05-11:
Thanks, I didn't read the instructions well enough. 👍 
I'm experienced with esp8266's and Arduino's but this is new to me...

When I select Board: "HoodLoader2 16u2", there is another thing that I can (must?) select:
Board / Bootloader / USB-Core:  
* "HoodLoader2 Uno"
* "HoodLoader2 Mega", 
* "HoodLoader2 Dev", 
* "DFU + USB-Serial Uno"
* "DFU + Serial Mega"
* "DFU"

In which case do I need to select which one?
(I have a Mega2560 with 16u2)

NicoHood, 2017-05-11:
This is just for burning the bootloader. Doesnt matter.

supersjimmie, 2017-05-11:
Thanks for explaining. Will close this now!

closed 2017-05-11 by supersjimmie