NicoHood/HoodLoader2 #51: 4 USB endpoints?

canadaduane, 2017-07-03:
In the Overview wiki page, it says:

```
4 usable USB Endpoints: CDC Serial, HID, MIDI (not all at once)
```

What does the "4" here refer to? Does one of the the 3 listed endpoints double as a something else?

NicoHood, 2017-07-03:
I dont understand the question, but CDC Serial uses 3 endpoints, HID 1 for each device (except multireport) and midi (i think) 2.

You can combine CDC + HID in most cases. That is the limitation.

canadaduane, 2017-07-03:
The confusion on my part is that the text says "4 endpoints" but then lists
only 3.

NicoHood, 2017-07-03:
I've corrected it and also fixed some headlines. Thanks.

canadaduane, 2017-07-03:
Thanks, that clarifies it for me. One other small thing: it's unclear if
the 500 bytes of RAM is what's left over after the hoodloader2 firmware is
installed?

```
500 bytes Ram 8u2 and 16u2 (1000 bytes Ram for a 32u2)
```

NicoHood, 2017-07-03:
https://en.wikipedia.org/wiki/Random-access_memory

HoodLoader2 is not a firmware, its a bootloader. And once it either loaded a new firmware or started the existing firmware it doesnt consume any ram. Thats how bootloaders work. Its the same for all arduinos, if it would be different i'd note it.

closed 2017-07-03 by NicoHood

canadaduane, 2017-07-03:
Ah, that makes sense. Thanks!