NicoHood/HoodLoader2 #52: Double-press reset doesn't enter bootloader mode?

canadaduane, 2017-07-16:
UPDATE: I just realized it's not the "RESET BUTTON" but the "RESET PIN" that needs to be double-shorted. Could that be clarified in the [How to use reset](https://github.com/NicoHood/HoodLoader2/wiki/How-to-use-reset) documentation (since there's a big button graphic up top that a lazy reader like myself could misunderstand?)

I've flashed the 16u2 with HoodLoader2, then flashed a compiled LUFA "Generic HID" demo. The USB enumerates for the demo, and all is well. However, there is no HoodLoader2 serial port enumerating (which is to be expected) and so I tried double-pressing the reset button to get out of LUFA and into HoodLoader2. The "L" LED flashes a bit, but nothing happens.

NicoHood, 2017-07-16:
There is no "button" word on the page and a big fat picture about the reset pin. How can I clarify this more?

So does it work for you now or not?

canadaduane, 2017-07-16:
It's the word "press" in the top image that threw me off. I'm used to the "reset" being a button--the idea of a reset pin is new (as I've learned about the 16u2 and ISP and so forth).

canadaduane, 2017-07-16:
Yes, I have it working.