NicoHood/HoodLoader2 #53: Can't install hoodloader2 on UNO r3

kGh05t, 2017-10-09:
at every 10 seconds led blink slowly then the 100ms blink again slow led......and it goes on....
using 100nf capacitor connected exactly as diagram....Don't know what is wrong..

NicoHood, 2017-10-11:
how is the 2nd usb mcu named? 16u2 or 8u2? do you have a 2nd arduino from which you can flash, so you can see debug messages?

kGh05t, 2017-10-12:
2nd usb mcu named as 16u2. Currently i haven't another arduino. That's why i'm using method A.

kGh05t, 2017-10-12:
i want to use my old ps/2 keyboard on my laptop. and if there any firmware for 16u2 so that arduino can be used as HID. let me know...

NicoHood, 2017-10-12:
I assume you connected something wrong. Have you checked if the 16u2 pinheader is flipped? I've mentioned it in the wiki also.

kGh05t, 2017-10-13:
i'll check it again and will let you know if that works.

kGh05t, 2017-10-13:
nah man....i didn't work...

kGh05t, 2017-10-13:
Just wanted to know if someone got it worked via  method A??

kGh05t, 2017-10-13:
if you have any firmware to use arduino as HID ..let me know ..Thanks.

closed 2017-10-16 by kGh05t