NicoHood/HoodLoader2 #60: Burn bootloader using ISP wiki needs update

dapug, 2018-04-13:
https://github.com/NicoHood/HoodLoader2/wiki/Burning-the-bootloader-via-ISP-%28advanced%29

This is about burning using an official ISP, but then proceeds to show how to burn via another Arduino which is already covered in another wiki page.   Also outdated, the Arduino IDE has been updated since this wiki was written.

I would be happyto submit a PR with updated docs, but I could not figure out how to use my ISP to get this thing up and running.  I have a AVRISP mkii and I have the Installation_Sketch loaded as per instructions, board set to Uno, and programmer set to mkii.  I get a generic error "Error while burning bootloader", with no further info from the IDE.  So that's it, don't know where to go from here.

NicoHood, 2018-04-14:
The wiki pages are correct. You can use your arduino as general ISP or via installation sketch burn another arduino. Its sounds similar, but its different. Burning via isp just works as usual, that nothing hoodloader specific. Also you can chose one of the other methods.

I will fix the headlines though.

mrbrainerpop, 2018-06-26:
I too get an error: "Error while burning bootloader" while trying to upload Hoodloader2. The factory bootloader flashes fine. Tried Flip (no joy), tried and failed with DFU Programmer as follows:

dfu-programmer.exe atmega16u2 erase --force
dfu-programmer.exe HoodLoader2_0_5_Mega_atmega16u2.hex --suppress-bootloader-mem --suppress-validation
dfu-programmer.exe atmega16u2 reset

Note: stk500boot_v2_mega2560.hex from the same Github download doesn't fail.

Also tried the ISP method. Same result, hoodloader2 download: stk500boot_v2_mega2560.hex works, hoodloader2 does not.

The problem seems to be that Hoodloader2 is 21kb in size. It simply doesn't fit into the mega2560? 

Tried the 8bit mega2560 hex file from the git file downloaded, it completes upload but then isn't recognized at all by Windows7. (To clarify: I have a real mega2560 not CH340 based MPU)

As a note: I own 4 Leonardos. I need allot of encoders on one platform otherwise, yes, I would use them or one of my many Pro's.

I'm only looking for help in making the above work as so many sites claim it does. If anyone has a possible information on why the above isn't working it would be greatly appreciated.

Perplexed...

NicoHood, 2018-06-27:
You are possibly flashing the wrong MCU. Please read the wiki carefully

closed 2018-06-27 by NicoHood