NicoHood/HoodLoader2 #62: LCD and AnalogRead Not Working

Glarity, 2018-06-18:
I had an LCD screen working before. Now it just shows filled in sqaures.
I am also reading buttons from the LCD keypad which also doesn't work anymore. Analogread(0) just returns 0 constantly. [This is what the keypad looks like for reference.](https://arduino-info.wikispaces.com/file/view/LCD-Pushbuttons-1-600.jpg/230299482/LCD-Pushbuttons-1-600.jpg)

NicoHood, 2018-06-23:
The 16u2 does not have Analog Pins at all. You are possibly flashing the wrong MCU. Please read the wiki carefully and try the examples first to understand the mechanisms better.

closed 2018-06-23 by NicoHood

Glarity, 2018-06-23:
Arduino Uno has 6 analog pins...

NicoHood, 2018-06-24:
I am talking about the 16u2, not the 328. Please read the wiki and the concept of the hoodloader2 carefully.