NicoHood/HoodLoader2 #64: Not compiling with visual micro/visual studio

Itsmetoo9, 2018-12-31:
When compilation is launched, (correct board) it throws 
```
cc1.exe*: error: -fno-fat-lto-objects are supported only with linker plugin
Error compiling core
```

The same is working with arduino IDE... Any Idea?

BlackOfWorld, 2018-12-31:
I have this issue too for a long time, I think it has something to do with visual micro itself.

NicoHood, 2018-12-31:
Your compiler has no LTO plugin. Please use avr-gcc from a linux distribution and the command line.

closed 2018-12-31 by NicoHood

Legion2, 2019-03-17:
Had same problem, solved it by deleting `...\AppData\Local\arduino15\packages\HoodLoader2\tools` folder, which contains old versions of the compiler. Now the newer version which comes with Arduino is used.