NicoHood/HoodLoader2 #67: Interrupt 1 (D14, PD1) only fires once directly after start on 32u2

M-Reimer, 2019-01-18:
The pin is listed as "Not connected" in your documentation, but the board, I'm trying to use, has it exposed.

I can read from Pin 14 without any issues and get the digital values. But for some reason the interrupt never seems to be called.

Is there any way to debug this?

  void setup() {
      Serial.begin(9600);
      pinMode(14, INPUT_PULLUP);
      attachInterrupt(digitalPinToInterrupt(14), isr, FALLING);
  }

  void isr() {
    Serial.println("int");
  }

NicoHood, 2019-01-26:
WTF that sounds strange. I have never tested the interrupts, so I guess you are right. I think this should not be patched here, but rather in the arduino core. The line you linked, where they swap the interrupts, could you please add another #ifdef atmega16u2 etc so that for those MCUs the issue will be fixed? That sounds like a better solution to me.

M-Reimer, 2019-01-26:
Do you think we actually have a chance to get this fixed in the "core" board files?
My guess would be, that they only support "their" boards and there never was any "official" Arduino with an 16u2 or 32u2 Atmel controller.

NicoHood, 2019-01-26:
They will fix it, they also integrated the 16u2 support. I had to patch some files in the core as well before. If the fix is correct, they will accept it. Give me the link and I will post a comment.

NicoHood, 2019-09-29:
This can be closed now, correct?

M-Reimer, 2019-09-29:
Once they release a new AVR core version. I would keep it open as the problem still exists for the average user.

M-Reimer, 2019-10-03:
FYI: https://github.com/M-Reimer/IwitVolumeKnob

The fact that the interrupt now depends on the AVR core version forces me to add some "hack":
https://github.com/M-Reimer/IwitVolumeKnob/blob/master/IwitVolumeKnob.cpp#L35-L60

NicoHood, 2019-10-13:
Well, you initiated that fix :-D

M-Reimer, 2019-10-13:
And you didn't want to mix in your code :-D

closed 2019-11-09 by NicoHood