NicoHood/HoodLoader2 #80: Odd behavior in 16u2

LoungeFlyZ, 2021-01-01:
I'm struggling to understand what might be happening when loading what looks like a relatively simple sketch on my megas 16u2. 

The code:

```
#include "HID-Project.h"

void setup() {
  Serial1.begin(9600);
  Keyboard.begin();
  // Start the USB Serial for debugging
  Serial.begin(9600);
  Serial.println("USB started");
}

void loop() {

  // the culprit:
  uint8_t my_str[10];
  my_str[0] = 'H';
  my_str[1] = 'e';
  my_str[2] = 'l';
  my_str[3] = 'l';
  my_str[4] = 'o';
  my_str[5] = 0;

  String str((char*)my_str);
  Serial.println(str);

  uint8_t str2[10];
  str2[0] = 'W';
  str2[1] = 'o';
  str2[2] = 'r';
  str2[3] = 'l';
  str2[4] = 'd';
  str2[5] = 0;
  
  String new_str((char*)str2);

  Serial.println(new_str);
  Serial.println(str + new_str);

  Serial.println("------");
  delay(5000); 
}
```

This results in:

```
Hello

------
```

However, if I comment out `Serial1.begin(9600);` from the setup, then I get the expected output. 

```
Hello
World
HelloWorld
------
```

I was attempting to build a simple bridge that receives messages from the i/o MCU and use your keyboard library to do things.  However I struck a problem with dealing with strings as shown above. 

I am hoping you might have some pointers given your experience with 16u2.  I am hoping its not something totally stupid that I've done :)

NicoHood, 2021-01-01:
Good question, this looks weird. The serial1 should not influence the usb serial (`Serial`). Does this also happens multiple times?

LoungeFlyZ, 2021-01-01:
Yes it happens every time. Comment out that line it works.  Uncomment, it stops working.

But here is the strange thing.  The Serial.println is printing an empty string with a `\n`.  That is why I get the blank line.   If I print the new_str length I get `0`.  

It's like using Serial1 breaks String after `Serial.println(str);`

NicoHood, 2021-01-01:
Can you try with normal string instead of the character array? I guess there is some weird c memory management stuff going on. I am not into that right now, doing a lot of other stuff recently. You may better ask in the arduino irc or the forum for help. I am not sure if this is related to this library. If you find the issue, please let me know.

LoungeFlyZ, 2021-01-01:
ok will do.  The reason for the char array is that is what I get from reading Serial1.  I was working on a simple bridge but needed to concat strings from serial1 until I got a full message.  I wanted to send

LoungeFlyZ, 2021-01-02:
Closing this issue.  @NicoHood it was a memory issue. During String concat (called when + was used) it was trying to realloc and failing because the  __malloc_margin is set to 128 and this didn't leave enough room.

Short answer is not to use String. 

Test work around is to use extern `size_t __malloc_margin;   __malloc_margin = 32;` and it all starts to work. 

Thanks to timemage on IRC for hunting and tracking this down.

closed 2021-01-02 by LoungeFlyZ

NicoHood, 2021-01-02:
Thanks for the feedback! I remember those guys, they are awesome! Helped me a lot too!

LoungeFlyZ, 2021-01-02:
Timemage remembered you too!