NicoHood/HoodLoader2 #81: Trying to build under VS2019 with VisualMicro Plugin

FHoevi, 2021-01-09:
I'm just trying to build the following code for a Mega2560 board on a Win10 box with Arduino IDE Ver 1.8.13 working in the background:

`void setup() {
   Serial.begin(115200);
}

void loop() {
   Serial.write(random('a', 'z'));
   delay(1000);
}`

During "Building core..." I get the following error message:
"C:\Users\XXX\AppData\Local\arduino15\packages\arduino\tools\avr-gcc\4.8.1-arduino5/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -flto **-fno-fat-lto-objects** -mmcu=atmega2560 -DVM_DEBUG_BREAKPAUSE -DVM_DBT=VM_DBT_GENERIC_OBJECT -DVM_DEBUGGER_SOFT_TRANSPORT_WRITER=Serial -DVM_DEBUGGER_SOFT_TRANSPORT=Serial -DVM_DEBUG_BANDWIDTH_THROTTLE_MS=50 -DVM_DEBUG -DVM_DEBUG_ENABLE=1 -DVM_DBT_GENERIC_OBJECT=21 -DVM_DBT_NO_SERIAL=20 -DVM_DBT_MS430_SERIAL_=19 -DVM_DBT_SERIALUSB=18 -DVM_DBT_USBAPI=17 -DVM_DBT_NET_UDP=16 -DVM_DBT_HARDWARESERIAL3=15 -DVM_DBT_HARDWARESERIAL2=14 -DVM_DBT_HARDWARESERIAL1=13 -DVM_DBT_CDCSerialClass=12 -DVM_DBT_COSA=11 -DVM_DBT_Uart=10 -DVM_DBT_NET_CONSOLE=9 -DVM_DBT_TTYUART=8 -DVM_DBT_USBSERIAL=7 -DVM_DBT_USART=6 -DVM_DBT_UART=5 -DVM_DBT_TEENSY=4 -DVM_DBT_USB=3 -DVM_DBT_FASTSERIAL=2 -DVM_DBT_SOFTWARESERIAL=1 -DVM_DBT_HARDWARESERIAL=0 -DF_CPU=16000000L -DARDUINO=108013 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -I"d:\Arduino\hardware\arduino\avr\cores\arduino" -I"d:\Arduino\hardware\arduino\avr\variants\mega" "d:\Arduino\hardware\arduino\avr\cores\arduino\hooks.c" -o "C:\Users\XXX\AppData\Local\Temp\VMBCore\arduino16x\ce7187fabc2083c01806427336f99201\core\hooks.c.o"
Error compiling core
Debug build failed for project 'BlaBla'

**cc1.exe*: error: -fno-fat-lto-objects are supported only with linker plugin**

When I take the same thing in the command line and skip the -fno-fat-lto-objects it seems to be working.

What do I have to do to get that working?

FHoevi, 2021-01-09:
OK, following [https://www.visualmicro.com/forums/YaBB.pl?num=1589058710/16](url) I had to delete the folder "C:\Users\XXX\AppData\Local\arduino15\packages\arduino". Now it uses avr-gcc from a different location which has been created during recent Arduino IDE 1.8.13 installation.

Problem solved.

closed 2021-01-09 by FHoevi

NicoHood, 2021-01-09:
Isnt this a general error you also get with the upstream arduino boards? It seems that the visualmicro plugin has a compiler that does not ship LTO. And the board definition file enables LTO:
https://github.com/arduino/ArduinoCore-avr/blob/master/platform.txt#L23

FHoevi, 2021-01-09:
When using the Arduino IDE anything was working well. I wanted to switch to VS2019 and VisualMicro for being able to set up larger projects and not just sketches. You are right, VisualMicro seems to to bring an outdated or not fully fledged compiler which needs to be removed by the step from above.

DRSDavidSoft, 2024-04-28:
Isn't this the same as #124?