NicoHood/HoodLoader2 #85: Found programmer: Id = "HL2.0.5"; type = S (everytime I tried to upload a program, This message come out and never get any use of the board after using your bootloader

mymafm0405, 2021-03-06:
Found programmer: Id = "HL2.0.5"; type = S (everytime I tried to upload a program, This message come out and never get any use of the board after using your bootloader

sw08, 2021-03-07:
Me too

I saw a post about this(Hoodloader2), and tried this.
But I think I coded right, but that message was printed every time, then my code doesn't work.

Here's the code.

```arduino
#include <LiquidCrystal_I2C.h>

LiquidCrystal_I2C lcd(0x27, 16, 2);

void setup() {
    lcd.init();
    lcd.backlight();
    pinMode(A0, INPUT);
    pinMode(A1, INPUT);
    lcd.clear();
    lcd.print("TwoHandle Mode");
    lcd.setCursor(0,1);
    lcd.print("Set Handles Left");
    lcd.setCursor(0,0);
    delay(5000);
    while (analogRead(A0) < 900 || analogRead(A1) < 900) {
        lcd.noBacklight();
        delay(500);
        lcd.backlight();
        delay(500);
    }
    lcd.clear();
    lcd.print("Start after 15 seconds.");
    delay(5000);
    lcd.clear();
    lcd.print("Start after 10 seconds.");
    delay(5000);
    lcd.clear();
    lcd.print("Start after 05 seconds.");
    delay(1000);
    lcd.clear();
    lcd.print("Start after 04 seconds.");
    delay(1000);
    lcd.clear();
    lcd.print("Start after 03 seconds.");
    delay(1000);
    lcd.clear();
    lcd.print("Start after 02 seconds.");
    delay(1000);
    lcd.clear();
    lcd.print("Start after 01 seconds.");
    delay(1000);
    lcd.clear();
    lcd.print("Start!");
    delay(1000);
    Serial.begin(115200);
}

void loop() {
    int Accel = 0;
    int Break = 0;
    lcd.clear();
    Accel = analogRead(A0);
    Break = analogRead(A1);
    Accel = round(4 - Accel / 205);
    Break = round(8 - Break / 114);
    String data = String(Accel) + "|" + String(Break) + ',';
    char Packet[data.length()];
    data.toCharArray(Packet, data.length());
    Serial.write(Packet);
    lcd.print("Accel: " + String(Accel));    
    lcd.setCursor(0, 1);
    if (Break != 8) {
        lcd.print("Break: " + String(Break));
        lcd.setCursor(0, 0);
    } else {
        lcd.print("Break: Emergency");
        lcd.setCursor(0, 0);
    }
    delay(100);
}``` - on 328P MCU

```arduino
#include <Keyboard.h>

int Break = 0;
int Accel = 0;

void setup() {
    Serial.begin(115200);
    Keyboard.begin();
}

void loop() {
    if (Serial.available()) {
        String data = Serial.readStringUntil(',');
        int sub = data.indexOf("|");
        int SerialBreak = data.substring(sub+1).toInt();
        data.remove(sub);
        int SerialAccel = data.toInt();
        if (true) {
            Break = 9;
            Keyboard.press('/');
            Keyboard.release('/');
        } else if (SerialBreak != Break) {
            if (SerialBreak < Break) {
                while (SerialBreak != Break) {
                    Keyboard.press(',');
                    Keyboard.release(',');
                }
            } else {
                while (SerialBreak != Break) {
                    Keyboard.press('.');
                    Keyboard.release('.');
                }
            }
        }
        if (SerialAccel != Accel) {
            if (SerialAccel < Accel) {
                while (SerialAccel != Accel) {
                    Keyboard.press('z');
                    Keyboard.release('z');
                }
            } else {
                while (SerialAccel != Accel) {
                    Keyboard.press('a');
                    Keyboard.release('a');
                }
            }
        }
    }
}``` - on 16u2 MCU

NicoHood, 2021-03-07:
I dont understand what your problem is. I suggest reading the wiki carefully, as most people dont understand, that they deal with 2 different MCUs now, and therefor need to use different pin and board options in the ide.

CerfontaineMathieu, 2021-03-18:
I think I missed something in the wiki. I read it about 5 times. 
Still can't read pin 2 on my uno while using HoodLoader. I got the same message but I don't know if it's related.
According to the schema in the wiki, only pin 13 and 3 should be unavailabe.
Can you clarify ?