NicoHood/HoodLoader2 #87: avrdude write/erase to USB MCU (using avr109) not working

JardaG, 2021-03-28:
I tried to use this tutorial today, but without success.
[Flash/erase firmwares(.hex files) with avrdude](https://github.com/NicoHood/HoodLoader2/wiki/Uploading-programs#flasherase-firmwareshex-files-with-avrdude)

HW:
- Arduino Uno R3
- USBasp (2011-05-28) for initial HootLoader2 HEX upload

SW:
- HoodLoader v2.0.5
- Windows 10 20H2 19042.867
- avrdude v6.3 for Windows
- Raspberry Pi OS 2021-03-04
- avrdude v6.3 for Linux

Windows device details:
```
USB Serial Device (COM31)
USB\VID_2341&PID_484C&MI_00\8&1277C7A7&0&0000
USB\VID_2341&PID_484C&REV_0100&MI_00
USB\VID_2341&PID_484C&MI_00
usbser.inf
21.06.2006
10.0.19041.1
HoodLoader2 16u2
\Driver\usbser
\Driver\usbccgp
```

Linux device details:
```
usb 1-1.1: new full-speed USB device number 28 using xhci_hcd
usb 1-1.1: New USB device found, idVendor=2341, idProduct=484c, bcdDevice= 1.00
usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
usb 1-1.1: Product: HoodLoader2 16u2
usb 1-1.1: Manufacturer: Arduino LLC
cdc_acm 1-1.1:1.0: ttyACM0: USB ACM device
```

Steps to reproduce (Windows):
- [Upload sketches to the USB MCU via IDE](https://github.com/NicoHood/HoodLoader2/wiki/Uploading-programs#upload-sketches-to-the-usb-mcu-via-ide).
- Try: `avrdude.exe -C avrdude.conf -p atmega16u2 -c avr109 -b 1200 -P COM31 -e`
- Error: `avrdude.exe: ser_drain(): read error: The I/O operation has been aborted because of either a thread exit or an application request.`

Notes:
- In Windows 10, the device will probably be disconnected immediately after starting avrdude (libusb0.dll provided), pls. see log.
[hootloader205_avrdude63_bug_win10.log](https://github.com/NicoHood/HoodLoader2/files/6216668/hootloader205_avrdude63_bug_win10.log)

- On the Raspberry Pi 4B, avrdude tries something, but the device does not respond, pls. see the log.
[hootloader205_avrdude63_bug_rpi4b.log](https://github.com/NicoHood/HoodLoader2/files/6216667/hootloader205_avrdude63_bug_rpi4b.log)

Is the error on my side or does it work for someone else?

NicoHood, 2021-03-28:
I am pretty sure it is on your side. Check the arduino ide. You can turn on verbose mode and check their command line of avrdude and see what the ide uses to flash a sketch to the 16u2. You can compare that with your command.

JardaG, 2021-03-28:
Using serial port named "Arduino Uno" COM30 immediately after fresh HEX HoodLoader2 upload I got this log and it is working, no problem here:
```
Forcing reset using 1200bps open/close on port COM30
PORTS {COM1, COM30, } / {COM1, COM30, } => {}
...
PORTS {COM1, COM30, } / {COM1, COM30, } => {}
Uploading using selected port: COM30
C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avrdude -CC:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf -v -patmega16u2 -cavr109 -PCOM30 -b57600 -D -Uflash:w:C:\Temp\arduino_build_393855/Blink.ino.hex:i

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf"

         Using Port                    : COM30
         Using Programmer              : avr109
         Overriding Baud Rate          : 57600
         AVR Part                      : ATmega16U2
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC6
         RESET disposition             : possible i/o
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
...
         Programmer Type : butterfly
         Description     : Atmel AppNote AVR109 Boot Loader

Connecting to programmer: .
Found programmer: Id = "HL2.0.5"; type = S
    Software Version = 1.0; No Hardware Version given.
Programmer supports auto addr increment.
Programmer supports buffered memory access with buffersize=128 bytes.

Programmer supports the following devices:
    Device code: 0x44

avrdude: devcode selected: 0x44
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% -0.00s
...
Writing | ################################################## | 100% 0.27s
...
Reading | ################################################## | 100% 0.03s

avrdude: verifying ...
avrdude: 3566 bytes of flash verified

avrdude done.  Thank you.
```

But then, if I use serial port "HoodLoader2 16u2" COM31, I got this error log:
```
Forcing reset using 1200bps open/close on port COM31
PORTS {COM1, COM31, } / {COM1, } => {}
PORTS {COM1, } / {COM1, } => {}
PORTS {COM1, } / {COM1, } => {}
...
PORTS {COM1, } / {COM1, } => {}
Couldn't find a Board on the selected port. Check that you have the correct port selected.  If it is correct, try pressing the board's reset button after initiating the upload.
```

I know that the speed of the 57600 serial port is for programming as stated in the documentation, and it works.
However, the speed of the serial port 1200 does not really work, the sketch cannot be loaded and the port is immediately disconnected from the system.

Although I'm able to switch to "Arduino Uno" with a quick double reset and use avrdude with AVR109 and speed 57600, it's not the procedure described in documentation with the 1200 speed, which just doesn't work.

What exactly should happen after opening the serial port at 1200? Can you provide me your logs please?

NicoHood, 2021-03-29:
1200 should reset the MCU as far as i know, so it will enter the bootloader mode. It turns out the reset does not work for you and you'd need to manually enter the bootloader mode. I currently also do not know why this happens to you.

JardaG, 2021-03-30:
After successfully uploading the sketch to the USB MCU, I remember the RX LED on the Arduino flashing quickly.
I think that fast flickering occurs even when using the speed 1200, but without HW I can't test it now.
I don't think I have an original Arduino Uno R3 but a clone.
I've checked the fuse bits before, but they were correct and I don't think it could affect, or not?