NicoHood/HoodLoader2 #88: raw hid and serial1 - work only if replace serial1 with register call

IgorIemeluyanofflyanov, 2021-04-17:
/*
  Copyright (c) 2014-2015 NicoHood
  See the readme for credit to other people.

  Advanced RawHID example

  Shows how to send bytes via RawHID.
  Press a button to send some example values.

  Every received data is mirrored to the host via Serial.

  See HID Project documentation for more information.
  https://github.com/NicoHood/HID/wiki/RawHID-API
*/

#include "HID-Project.h"

const int pinLed = LED_BUILTIN;

// Buffer to hold RawHID data.
// If host tries to send more data than this,
// it will respond with an error.
// If the data is not read until the host sends the next data
// it will also respond with an error and the data will be lost.
uint8_t rawhidData[64];

const long  FOSC = 16000000L; // Clock Speed
const int BAUD = 9600;
const int MYUBRR = (FOSC / 16 / BAUD - 1);

void USART_Init( )
{
  /*Set baud rate */
  UBRR1H = (unsigned char)(MYUBRR >> 8);
  UBRR1L = (unsigned char)MYUBRR;
  UCSR1B |= (1 << RXEN1) | (1 << TXEN1);   // Turn on the transmission and reception circuitry

}

void USART_Transmit( unsigned char data ) {
  /* Wait for empty transmit buffer */
  while ( !( UCSR1A & (1 << UDRE1)) )
  {
    ;
  }
  /* Put data into buffer, sends the data */
  UDR1 = data;
}

char USART_Receive( void )
{
  /* Wait for data to be received */
  while ( !(UCSR1A & (1 << RXC1)) )
    ;
  /* Get and return received data from buffer */
  return UDR1;
}

void setup() {
  pinMode(pinLed, OUTPUT);

  USART_Init();

  // Set the RawHID OUT report array.
  // Feature reports are also (parallel) possible, see the other example for this.
  RawHID.begin(rawhidData, sizeof(rawhidData));
}
uint8_t i = 0;
uint8_t megabuff[sizeof(rawhidData)];
void loop() {

  digitalWrite(pinLed, HIGH);

  // Create buffer with numbers and send it

  if  ( UCSR1A & (1 << RXC1))
    megabuff[i++] =  UDR1;// Serial1.read();

  if (i == sizeof(megabuff))
  {
    RawHID.write(megabuff, sizeof(megabuff));
    i = 0;
    //    delay(300);
  }

  // Simple debounce

  digitalWrite(pinLed, LOW);

}

NicoHood, 2021-04-17:
What is your problem? This does not seem related to hoodloader2.

IgorIemeluyanofflyanov, 2021-04-17:
Nico, great thank for your job ! Its very usefully. I can't make combination RAW HID + Serial1 for link Uno with PC by HID instead Serial. If remove Serial1 from sketch and replace by UART then work..

NicoHood, 2021-04-17:
I am not sure, I think rawhid does not work in general. I am not sure why it even works with your solution