NicoHood/HoodLoader2 #106: MacOS Monterey: bad CPU type in executable

TheBlackfurGuy, 2022-06-24:
Heyho, I tried to setup HoodLoader2 + HID-Project and the installation went alright but if I now try to compile an example sketch I get this error in the output log:
`Compilation error: fork/exec /Users/blackfur/Library/Arduino15/packages/arduino/tools/avr-gcc/4.8.1-arduino5/bin/avr-g++: bad CPU type in executable`

The issue seems to be that avr-gcc in version 4.8.1 is not available as a x86_64 binary and thus cannot be executed on any Mac running MacOS 10.15 or higher.

Do you have nay idea what I could try?

NicoHood, 2022-06-25:
Maybe the board file needs an update? Does arduino use newer versions in the meantime?

I just had a quick look, but I cannot see that this error comes from this repository. Does compiling for other devices work for you?

TheBlackfurGuy, 2022-06-25:
> Maybe the board file needs an update? Does arduino use newer versions in the meantime?

This probably, arduino currently uses `7.3.0-atmel3.6.1-arduino7` with the vanilla Uno board.

> I just had a quick look, but I cannot see that this error comes from this repository. Does compiling for other devices work for you?

Well the issue is that arduino doesn't offer a 64 bit version of 7 year old compiler. Yes compiling for other devices works.

I know the issue, (since this is a common bs problem on macOS) but I know too little about arduino IDE board files to do anything about it.

TheBlackfurGuy, 2022-06-25:
Another quick look the issue lies in this file: `https://raw.githubusercontent.com/NicoHood/HoodLoader2/master/package_NicoHood_HoodLoader2_index.json` under `toolsDependencies` there is the entry `"version": "4.8.1-arduino5"`

NicoHood, 2022-06-25:
Thanks, that helped.

@per1234 Could you please have a look? I think you have helped me packaging this, can we update it?

TheBlackfurGuy, 2022-06-25:
I did it, have a look at the most recent release :D https://github.com/TheBlackfurGuy/dumpster-fire/releases
Just replaced the old version with the one the most recent version of the arduino board definition uses.

TheBlackfurGuy, 2022-06-25:
Everything works wonderfully now. I'll make sure to buy you a coffee :)

per1234, 2022-06-25:
The version numbers of the `arduino:avr-gcc` and `arduino:avrdude` tool dependencies selected by @TheBlackfurGuy are the ones used in the most recent release of the official "Arduino AVR Boards" platform, so I think updating to them is a good idea.

However, I don't think the specific change made in https://github.com/TheBlackfurGuy/dumpster-fire/releases/download/dumpster-v3/package_NicoHood_HoodLoader2_index.json (modifying the existing HoodLoader2 2.0.5 release is the right approach. The correct approach is to make a new release. The most concrete reason is that this will push the update out to all the existing users via the Arduino Boards Manager system, which would not happen if you only modify the existing release. In addition, it is really not good development practices to make changes to an existing release. If you don't have any  changes on the platform files themselves you want to release, you could simply reuse the existing content of the 2.0.5 release, with the only change being the updated tool dependency versions.

These are very significant bumps, so some validation would be a good idea. Offhand, I can't think of 
any issues to be concerned about, but definitely worth at least running some "smoke test" compilations  and uploads to make sure nothing is completely broken.

TheBlackfurGuy, 2022-06-25:
Yeah it was just a quick test to see if it even works or if gcc broke for 16u2 in the meantime.

NicoHood, 2022-06-28:
@per1234 Thanks for those tips! Could you help me package the zip file that must be uploaded in order to make the board manager compatible? I think we need to bump the versions via pr, tag a new release and then attach the zip files? I dont remember good how it is done.

per1234, 2022-07-02:
## Make release

The GitHub releases system is used because it allows attaching "assets" files to a Git tag, which is used for distribution of the release archive rather than having to check it into the repository.

1. Decide the version number of the new release.
1. Update the `version` value in `platform.txt` to the new version number:
   https://github.com/NicoHood/HoodLoader2/blob/0c1c4bbf338383cb02757ad965f1dc50657ba984/avr/platform.txt#L9
1. Open the GitHub release creation page:
   https://github.com/NicoHood/HoodLoader2/releases/new
1. Click on the "**Choose a tag**" menu.
1. Type the version number into the field.
1. Select "**+ Create a new tag: \<tag name\> on publish**" from the dropdown menu.
1. Click the <kbd>**Publish release**</kbd> button.

## Make release archive

This is the file containing the platform release, which will be downloaded by the Arduino Boards Manager.

1. Make the release archive:
   ```text
   TAG_NAME="<tag name>"
   git clone --branch $TAG_NAME https://github.com/NicoHood/HoodLoader2
   cd HoodLoader2/
   ARCHIVE_NAME="${TAG_NAME}-boards_manager.zip"
   zip -9 -r "$ARCHIVE_NAME" avr/
   ```
1. Open the GitHub release page:
   https://github.com/NicoHood/HoodLoader2/releases/latest
1. Click the pencil icon near the top right corner of the release page.
1. Click on "**Attach binaries by dropping them here or selecting them.**".
1. Select the release archive file you created.
1. Click the <kbd>**Open**</kbd> button.
1. Click the <kbd>**Update release**</kbd> button.

## Make package index entry

This is [the data file](https://arduino.github.io/arduino-cli/latest/package_index_json-specification/) Arduino Boards Manager uses for platform installations and updates.

1. Determine the checksum of the release archive:
   ```text
   sha256sum "$ARCHIVE_NAME"
   ```
1. Determine the size of the release archive:
   ```text
   stat --printf="%s" "$ARCHIVE_NAME"
   ```
1. Write a package index entry for the release:
   ```json
   {
     "name": "HoodLoader2",
     "architecture": "avr",
     "version": "TODO_TAG_NAME",
     "category": "Contributed",
     "help": {
       "online": ""
     },
     "url": "https://github.com/NicoHood/HoodLoader2/releases/download/TODO_TAG_NAME/TODO_TAG_NAME-boards_manager.zip",
     "archiveFileName": "TODO_TAG_NAME-boards_manager.zip",
     "checksum": "SHA-256:TODO_CHECKSUM",
     "size": "TODO_SIZE",
     "boards": [
       {"name": "16u2"},
       {"name": "32u2"},
       {"name": "8u2"},
       {"name": "at90usb82"},
       {"name": "at90usb162"},
       {"name": "32u4"},
       {"name": "Uno"},
       {"name": "Mega 2560"}
     ],
     "toolsDependencies": [
       {
         "packager": "arduino",
         "name": "avr-gcc",
         "version": "7.3.0-atmel3.6.1-arduino7"
       },
       {
         "packager": "arduino",
         "name": "avrdude",
         "version": "6.3.0-arduino17"
       }
     ]
   }
   ```
1. Add the entry to the `packages.platforms[]` key in the package index file:
   https://github.com/NicoHood/HoodLoader2/blob/master/package_NicoHood_HoodLoader2_index.json
1. Commit and push the package index entry to the GitHub repository.

hraftery, 2023-11-28:
Specifying the link in @per1234 's [comment](https://github.com/NicoHood/HoodLoader2/issues/106#issuecomment-1166249286) as the "Additional boards manager URLs" resolved the error for me (noting however, that that's not why `per1234` provided it!).

For me the error appeared when I tried to use Arduino IDE to Flash the bootloader using Arduino as ISP, on MacOS Sonoma 14.1.1 running on an M1, with the same symptoms described in this issue.

DasBlackfur, 2024-04-12:
@hraftery You are welcome :D