NicoHood/HoodLoader2 #109: Image in Hoodloader wiring is a bit old?

FrenzyExists, 2022-07-30:
I don't know if this is still being worked on, but here goes nothing. I wanted to use my UNO to control my laptop, so I got the HID-Project on PlatformIO. It didn't compile, since UNO seems to be limited in its USB capabilities. After some research I found this repo, and looking the guide, I noticed that the UNO there is the older versions. My UNO is the R3 model, only has 1 set of ICSP pins

[image]

but on the picture, there's two sets. 

[image]

So, how would the wiring be for R3?

NicoHood, 2022-07-30:
No, it is not old, you just bought a cheaper china clone. It does not have the USB MCU, just a cheap usb-serial chip.

closed 2022-07-30 by NicoHood

FrenzyExists, 2022-07-30:
I guess that means I won't be able to install hootloader. That sucks :\