NicoHood/HoodLoader2 #119: Can't compile anything using HootLoader2 boards

svilex, 2022-12-28:
I installed HootLoader2 on my arduino UNO R3. Also verified the install, all good. I'm still able to normally use it as an UNO as expected:
[image]

The problem is, whenever I switch board to any of the HootLoader2 boards and try to compile anything, it always fails with the same error:
[image]

What am I doing wrong?

NicoHood, 2022-12-29:
I guess it is not compatible with the new arduino ide. I've never tested that, as arduino does not want to help me build the ide on linux.

svilex, 2022-12-29:
Also not working on windows, different compiler error though. Legacy IDE works fine.

biologyscience, 2023-06-13:
> Also not working on windows, different compiler error though.

Hey @svilex if you are still here, was the error about compiling or some error which occurs while uploading the sketch.
Because I get an error while uploading, just wanted to confirm before going to the legacy IDE

DRSDavidSoft, 2024-08-18:
@NicoHood Let's move away from the Arduino IDE which as this point is just the [Theia IDE](https://theia-ide.org/) running the [Monaco](https://microsoft.github.io/monaco-editor/) editor.

Does the new Arduino CLI work with HoodLoader2? Now that would be awesome.

martian17, 2026-02-08:
I tried building with arduino-cli, but I see the same error as what I see on IDE
```bash
$ arduino-cli compile --fqbn HoodLoader2:avr:HoodLoader2atmega16u2:board=uno .
/home/xxx/.arduino15/packages/arduino/hardware/avr/1.8.7/cores/arduino/WString.cpp:36:47: error: 'DECIMAL_DIG' was not declared in this scope
 size_t const String::FLT_MAX_DECIMAL_PLACES = DECIMAL_DIG;
                                               ^
/home/xxx/.arduino15/packages/arduino/hardware/avr/1.8.7/cores/arduino/WString.cpp:37:47: error: 'DECIMAL_DIG' was not declared in this scope
 size_t const String::DBL_MAX_DECIMAL_PLACES = DECIMAL_DIG;
                                               ^
/home/xxx/.arduino15/packages/arduino/hardware/avr/1.8.7/cores/arduino/WString.cpp: In constructor 'String::String(float, unsigned char)':
/home/xxx/.arduino15/packages/arduino/hardware/avr/1.8.7/cores/arduino/WString.cpp:127:25: error: size of array 'buf' is not an integral constant-expression
  char buf[FLOAT_BUF_SIZE];
                         ^
/home/xxx/.arduino15/packages/arduino/hardware/avr/1.8.7/cores/arduino/WString.cpp: In constructor 'String::String(double, unsigned char)':
/home/xxx/.arduino15/packages/arduino/hardware/avr/1.8.7/cores/arduino/WString.cpp:136:26: error: size of array 'buf' is not an integral constant-expression
  char buf[DOUBLE_BUF_SIZE];
                          ^

Benutzte Bibliothek Version Pfad (Using library version path)
IRremote            4.5.0   /home/xxx/Arduino/libraries/IRremote

Verwendete Plattform Version Pfad (Using platform version path)
HoodLoader2:avr      2.0.5   /home/xxx/.arduino15/packages/HoodLoader2/hardware/avr/2.0.5
arduino:avr          1.8.7   /home/xxx/.arduino15/packages/arduino/hardware/avr/1.8.7
Fehler beim Build: exit status 1 (Build error: exit status 1)
```
OS: Arch Linux x86_64

djrcapo, 2026-02-08:
@martian17 The problem is due to the IDE version, using Legacy version will work fine.