NicoHood/HoodLoader2 #124: Error compiling program for HoodLoader2 Uno

Sanjeet990, 2023-01-29:
I was able to flash HoodLoader2 correctly. I can also veryfy this in devices and printers but while flashing any program, I'm getting this:

cc1.exe: error: -fno-fat-lto-objects are supported only with linker plugin.
cc1.exe: error: -fno-fat-lto-objects are supported only with linker plugin.
cc1.exe: error: -fno-fat-lto-objects are supported only with linker plugin.
cc1.exe: error: -fno-fat-lto-objects are supported only with linker plugin.
cc1.exe: error: -fno-fat-lto-objects are supported only with linker plugin.
cc1.exe: error: -fno-fat-lto-objects are supported only with linker plugin.
cc1.exe: error: -fno-fat-lto-objects are supported only with linker plugin.

exit status 1

Compilation error: exit status 1

Bnco3DTech, 2023-12-04:
Did you fix it I have same issue.

cc1.exe: error: -fno-fat-lto-objects are supported only with linker plugin.

Kandermann, 2024-01-17:
I am having this issue too!
kind of wierd

TannerGilbert, 2024-02-09:
I also got the same error when using Arduino IDE 2.x. Using Arduino IDE 1.8.x instead solved the problem for me.

DRSDavidSoft, 2024-04-28:
@NicoHood Same issue with VS Code and the Arduino extension :(

[image]

syfenx, 2024-07-10:
pretty sure I fixed this by installing 1.8.3 Arduino AVR Boards in board manager

SlopeSlayer910, 2024-08-14:
> Using Arduino IDE 1.8.x instead solved the problem for me.

I am also encountering this error. Switching to Arduino IDE 1.8.19 solved the problem for me as well however compatibility with Arduino IDE 2.X.X would be greatly appreciated.

Sanjeet990, 2024-08-18:
The issue above fixes by installing the proper board definitions in board manager. Closing it to prevent any confusion.

closed 2024-08-18 by Sanjeet990

DRSDavidSoft, 2024-08-18:
@Sanjeet990 Are you saying that the issue will not occur with Arduino 2.X as well? It was established that the legacy IDE doesn't have this issue anymore.

Also I use VS Code with the [Arduino extension by Microsoft](https://marketplace.visualstudio.com/items?itemName=vsciot-vscode.vscode-arduino) which caused the error I posted.

For building the HoodLoader I would prefer not to **need** the legacy IDE since everyone is switching away from it to the new, modular, cool v2.x IDE. Would you please re-open the issue until this is also fixed for the v2.x?