NicoHood/HoodLoader2 #126: FLIP Gives "Address out of range" for Arduino UNO

HMI1973, 2023-02-12:
When I try to flash the bootloader to Arduino UNO it gives me error "Address out of range" using 16U2
Any idea how to fix that issue?

NicoHood, 2023-02-12:
You have a 8u2 on the board maybe?

NicoHood, 2023-02-12:
Oh no, you cannot upload it using flip. It is a bootloader, not a firmware. Please read the wiki carefully

closed 2023-02-12 by NicoHood

HMI1973, 2023-02-13:
Thank you for your fast reply i didn't expect to have this fast answer to my question

First i use Arduino Uno R3 the chip says it's 16U2 but when I use flip it gives Address out of range.
when i tried 8U2 it goes correctly and FLIP could burn it but unfortunatly even i upload the HOODLOADER2 for 8u@ the device can't be recognized as HID and it gives error "no response descriptor from the device".

As I understand from you reply I can't use FLIP as bootloader burner, can I know more details about this and what is the proper tool/Hardware I can use to burn bootloader? should i use ISP board like USBASP or use Arduino as ISP?

Finally I have assembly background but new in command line compile of Arduino codes, I need to modify the bootloader myself to change some HID descriptor configuration. can you support me with resources for bootloader source code and how to compile it?

Thank you again for your great support.

Regards.

NicoHood, 2023-02-13:
https://github.com/NicoHood/HoodLoader2/wiki/Hardware-Installation